jgyates/genmon #322: USB Port - inquiry

mwarchut-monsoon, 2019-11-06:
My new generator has USB port.   That can be used instead of the serial cable?  If not what is it for?

swvalenti194, 2019-11-06:
That's to push firmware updates to controller.

mwarchut-monsoon, 2019-11-07:
So I either may not have the port or its hidden under all that metal.?  It's not readily visible.

jgyates, 2019-11-07:
model number and pictures may help give us a reference to help you better. Typically the connector with the serial port are on the back of the controller. sometimes the connector may be covered with a sticker. If you have an Evolution 2 then this is the connector that the wifi module is plugged into. If you have  a Nexus or Evolution 1 then the connector is unused (unless you use Mobile Link)

mwarchut-monsoon, 2019-11-07:
Its an Evolution 2.0 controller.  I have the IP address that it is using. Can the controller be accessed over IP?

liltux, 2019-11-07:
@mwarchut-monsoon I'm sure something is accessable via the IP address since that is how Big G is pushing updates to the controller.  As too which protocol or how to access I have not heard of anything.  Use Wireshark or other network sniffing utilities and find out.

mwarchut-monsoon, 2019-11-07:
Good thing I am a network engineer.. I have all the cool tools..   :)

mwarchut-monsoon, 2019-11-07:
Actually I would bet that the firmware update is a pull request from the controller..  Firewalls properly configured would not allow Generac to initiate that.

mwarchut-monsoon, 2019-11-07:
A scan revealed that all ports except these are closed.  The ones that are in use are most likely protected by a firewall..

Not shown: 65521 closed ports
PORT      STATE    SERVICE
1989/tcp  filtered tr-rsrb-p3
12757/tcp filtered unknown
20301/tcp filtered unknown
20793/tcp filtered unknown
25522/tcp filtered unknown
25756/tcp filtered unknown
28213/tcp filtered unknown
35408/tcp filtered unknown
43084/tcp filtered unknown
43648/tcp filtered unknown
52355/tcp filtered unknown
58900/tcp filtered unknown
60471/tcp filtered unknown
61085/tcp filtered unknown

jgyates, 2019-11-08:
Since the generator is initialing the request a proxy server would need to be setup to be the man in the middle between the generator and generac servers in any solution.  I would be happy to incorporate this into the project. I just don't have access to an Evolution 2 controller.

Also, this thread has some pictures of the Evolution 2 connector and the thread also has some links to other threads with more pictures.

https://github.com/jgyates/genmon/issues/274

jgyates, 2019-11-10:
Closing this issue for now. Feel free to post to this thread if you have  other questions on this topic.

closed 2019-11-10 by jgyates