jgyates/genmon #323: Doesnt detect mode number, and incorrectly says it is propane.

scyto, 2019-11-09:
Use the template below if you have an issue or want to report a bug. If you have a question or a feature request you can ignore the questions below.

***NOTE:*** If you are having issues with your serial connection, please read this page before posting:

**### Expected Behavior**

should show model number and that it is NG not propane
model is 0058750 and it is NG

**### Actual Behavior**

Actually shows:

Maintenance :
    Model : Unknown
    Generator Serial Number : 0006109036
    Controller : Nexus, Air Cooled
    Nominal RPM : 3600
    Rated kW : 20 kW
    Nominal Frequency : 60 Hz
    Fuel Type : Propane
    Engine Displacement : 999 cc

### Screenshot or Pictures relating to the problem (if possible)

{Please write here}

### Your Environment

- Generator Model: 0058750
- Generator Registers: Submit via the About page - done
- Genmon Version: v1.13.30
[image]

scyto, 2019-11-09:
for the propane / natural gas i now realize it can't detect and it is a setting I set in the UI.

Model number still not detected, unless i am being silly on that one too.

jgyates, 2019-11-09:
Yes, the fuel selection will only work if the model is detected.

The way model detection works is this:

     The serial number is used to look up the model number from generac's web site. You must be connected to the internet for this to work. The look up, if successful will only be done once unless the version if changed, then  it will be performed again. Manually looking up your serial number appears to work using this link:

    http://www.generac.com/service-support/product-support-lookup

You can send your logs files from the about page and that should let me know if this is an error in the program or if the software was unable to reach the generac web site.

scyto, 2019-11-09:
@jgyates thanks for the reply, i did click the send logs and registers when i submitted the ticket, it it uses email it won't have worked cause i only just got email working (MFA on google and o365 royal PITA lol) - i can resend.

jgyates, 2019-11-09:
I got your logs. I don't see any errors related to the lookup. I do some serial errors that were logged. One thing that you can do is this:

1) Log into the pi via ssh
2) edit the file /etc/genmon/genmon.conf

         sudo nano /etc/genmon.conf 

3) remove the line that looks like this, is should be near the bottom of the file

        version = V1.13.30

4) save the file (Ctrl+X, then Y, then Enter)

5) restart genmon
     
       cd /home/pi/genmon
       bash ./startgenmon.sh restart

This will make the model look up occur again. If your cable was not connected when genmon first loaded or if you got a serial error on the first load, this could cause this problem and removing the version from the conf file will make genmon thing this is a new install and do the look up again.

scyto, 2019-11-09:
yes that worked, with the caveat i think you meant
```
 sudo nano /etc/genmon/genmon.conf 
```
yes when i first started genmon no cables were attached, that came several days later....

closed 2019-11-09 by scyto