jgyates/genmon #324: Web Server Stops Responding

restauffer, 2019-11-10:
Use the template below if you have an issue or want to report a bug. If you have a question or a feature request you can ignore the questions below.

***NOTE:*** If you are having issues with your serial connection, please read this page before posting:

https://github.com/jgyates/genmon/wiki/3.6---Serial-Troubleshooting

If you are having other issues, please see the following page:

https://github.com/jgyates/genmon/wiki/3.5---General-Troubleshooting

If you need to send you logs and registers to the developer, if you email is setup and working properly you can click send your logs on the About page in the web interface.

### Expected Behavior

Genmon will continue to be available without regular restarts

Every few days the web server will stop responding. If I use ClientInterface.py I can see genmon is still running and receiving data from the generator. I am running the SSL server with a Cert from Certbot. I cannot find anything in any log to indicate the cause.

{Please write here}

### Steps to Reproduce (including precondition)

Wait several days and it will always occur

### Screenshot or Pictures relating to the problem (if possible)

{Please write here}

### Your Environment

- Generator Model: G0070421, Evolution, Air Cooled
- Generator Registers: Submit via the About page
- Genmon Version: V1.13.29

jgyates, 2019-11-10:
This problem sounds a lot like one I saw before in this thread:

https://github.com/jgyates/genmon/issues/271

The solution was related to their router and port forwarding / DDNS issue.

Are you using port forwarding? if so who makes your router?

restauffer, 2019-11-10:
Thanks for the reply. I read that thread before but just re-read it more thoroughly. Some points:

- I am using a Ubiquiti Router and  I am using port forwarding. This problem has been occurring almost since I started using Genmon over a year ago. I was using a Syslink router with DD-WRT on it before the Ubiquti and had the same problems.
- At one point I turned off port forwarding thinking the web server might be getting overloaded with portscans, etc. I did not make any changes to Genmon. It did not fix the problem. 
-  Unlike the poster in 271, my instance normally fails only after several days. 
-  I am using a public cert from Certbot
- I get certificate warnings with Chrome as the cert is registered to my external DDNS name, not to the internal IP.
- Genmon itself stays up when the problem occurs. I can verify this with ClientInterface.py
- The PI itself is running fine when this happens. I can SSH to it or use VNC to get to the GUI.
- The problem occurs whether attempting to connect via the external IP (via port forwarding) or on the internal IP of the PI.

Thanks

jgyates, 2019-11-10:
with the data  you provided, it sounds like the issue may be  related to the Flask library internal web server when using a your cert. 

There are  a few ways you can go with this:

 1) continue troubleshooting to try to nail down the exact cause, possibly run a test with a cert with the IP of the pi with no forwarding to see if the web server will fail. If the problem ends up being an issue with the Flask library then move on to 2 or 3 below. Presently,  I do not know of anything I can change in genserv.py that would affect this behavior.
2) Use a VPN instead of exposing the web server  to internet. The makers of the Flask library do not recommend this since the internal web server is built for single use (i.e. one user at a time). This is how I use the web interface outside of my local network. I have used the built in Ubiquiti VPN server with the built in Android, MacOS and Windows VPN clients. No cert required and no outside ports are open.
3) Setup apache with Flask to serve the web interface on an exposed port instead of using the internal Flask webserver. There are many articles on how to set this up on the web (for example https://www.codementor.io/abhishake/minimal-apache-configuration-for-deploying-a-flask-app-ubuntu-18-04-phu50a7ft). I have not taken the time to incorporate this into the project since the VPN method is working well for me and is more secure.

I hope this helps. Let me know if you have any other questions.

restauffer, 2019-11-14:
Thanks. I will set up some tests. No HTTPS, HTTPS without port forwarding, Cert on internal address, etc. and see if narrow down the culprit. Since my server often runs for close to a week before stopping it may take some time to complete the tests. If I find something, I will post it here.

jgyates, 2019-12-20:
I am going to close this thread but feel free to post any follow up data or questions to this thread.

closed 2019-12-20 by jgyates

restauffer, 2019-12-20:
Still testing. Results so far:

1 HTTP - No NAT redirects - Ran 10 days with no issues
2 HTTP -Nat redirect to port 443 on PI (even though PI not listening on Port 443) - Ran 10+ days with no issues.
3. HTTPS - Cert on external IP, no NAT redirect to PI - Failed at ~8 days.

I will continue tests but at some point I'll just consider a CRON job that restarts the server one each day. It always takes days before it fails so that would be a workaround

pburger1, 2020-11-04:
@jgyates  -- I discovered GenMon in August 2020, after my generator controller got fried in a lightning storm.  What a great project! -- thanks.

I am having this same issue with the webserver stopping after about a week or so.  At first I thought my wifi adapter had quit, and replaced it, but later realized that wasn't the problem.  I still get the emails from GenMon when the generator exercises each Wednesday, so I'm not losing WIFI connectivity (although my WIFI signal is very low).  I can reset the GenMon and the webserver starts working immediately and stops again after a week or so. 

Any chance someone could help with setting up "a CRON job that restarts the sever each day".  I'm an engineer, but not a python programmer.  I'll need some hand-holding for this.

jgyates, 2020-11-04:
To setup a cron job type this to edit the crontab file:

     sudo crontab - e

sudo is for admin (root) rights. You may get a prompt asking which edit to use. Select nano as the editor.

This will bring up a text file, it should have the line that starts genmon on boot. Below that line enter.

      0 22 * * * /bin/bash /home/pi/genmon/startgenmon.sh restart

Then save the file by typing Ctrl+X, then Y, then Enter.

The 22 in the line above is the 22nd hour (10pm). This page has more info:

     https://ostechnix.com/a-beginners-guide-to-cron-jobs/

Note: the above command will restart genmon and genserv (the web server).  Since both programs are loaded with genloader this will keep things working as expected since the process ID is stored by genloader. If you only restarted the web server then using the command startgenmon.sh restart will not work as expected. If you are OK with this then the command to only start the web server would be to add this line to the crontab instead:

       0 22 * * * sudo pkill -9 -u root -f genserv.py
       1 22 * * * sudo python /home/pi/genmon/genserv.py -c /etc/genmon/

This will kill genserv at 10:00pm and start it again at 10:01 pm, but with the caveat mentioned above.

If you are having issues, try disabling any NAT or port forwarding. I don't use port forwarding but prefer a VPN. Most routers support VPN access.

pburger1, 2020-11-04:
@jgyates Thanks so much for the quick reply.  I will probably try the first option and see how it goes.  I'll probably do this over the weekend.  I am currently using port forwarding, but may switch to VPN at some point in the future.  I'll report back with the results in a week or two for the benefit of anyone else having similar issues.  I suspect this will solve the problem.

restauffer, 2020-11-05:
The problem is definitely related to using certificates. It does not matter
if you leave port 80 open or blocked 24/7 and the failure does not seem to
coincide with the cron job that checks if a new cert is needed. I have
never found anything in the logs. I have determined at what time it stops
by leaving a window open on the  The only that I have found  stops the
problem is not using SSL and going the VPN route instead.

I took the route of keeping ssl and using  the cron job. I found, as
jgaytes indicated, a restart leaves things not working. I do a stop and
then 1 minute later a start which is the same thing jgaytes is recommending.