jgyates/genmon #326: emails not sending shortly after startup / unknown 0x08 stuck in logs

tingraham1, 2019-11-16:
### Expected Behavior

E-mails should send on engine state changes/exercises. 

### Actual Behavior

After rebooting PI or installing updates which restarts services I get the 2 emails and then none after that.  I can send test notifications just fine and this originally was working fine until it now doesn't.  Also have this in the alarm log for 4 months "Unknown 0x08".  I have a Nexus controller.

### Steps to Reproduce (including precondition)

Explained above

- Generator Model: 2014 Generac 14KW Nexus
- Generator Registers: Submit via the About page
- Genmon Version: V1.13.31

tingraham1, 2019-11-16:
Annnnd just after posting I think I sorted the mail issue.  I had automatic feedback on and because I am using an internal mail server for relay it can't send to an external recipient and because I have that "unknown 0x08" in the logs the email is constantly trying and failing to send to your email.  I turned the automatic feedback and it appears to be sorted.  Now, any ideas on the 0x08?  thanks!

jgyates, 2019-11-16:
If the unknown value is in the alarm log,  would need to know the  alarm type in your log. If you can look at the time and date of the genmon alarm log entry in question and compare it to the corresponding alarm log read from the controller via the controller display,  then give me the exact text for that alarm entry, I can add it to the software.

tingraham1, 2019-11-16:
Just checked it and it is "WIRING ERROR" triggered from when I unhooked the generator whip several months ago.

jgyates, 2019-11-16:
I just  checked in V1.13.32. This should fix your issue.

tingraham1, 2019-11-16:
You are the best!  Thanks a ton!

closed 2019-11-16 by tingraham1