jgyates/genmon #327: Genmon unable to start/stop generator

klinquist, 2019-11-19:
Hi,
I have a Honeywell/Generac 20kW/Evolution 2.0.  Running a GenPi purchased on eBay.

I can see the generator status, but for some reason the stop / start / start & transfer options are not working.   Any ideas how to troubleshoot?

Thanks!

jgyates, 2019-11-19:
The Monitor page in the web interface should be the first place to get info about the health of the system, specifically the Generator Monitor Stats, Serial Stats and Platform Stats.  A posted screenshot of this page may help. I assume the generator has been registered with generac and is otherwise functional (unregistered gensets will not start when the power is lost  and possibly on remote  commands).

I also assume you are using a transfer switch (i.e. you are NOT using two wire start)  as the remote commands may not work with two wire mode.

If anyone with an Evo2.0 setup can chime in and validate that the remove start/stop/transfer functionality works with Evo 2.0 that would be helpful also.

liltux, 2019-11-19:
@klinquist is your unit in Auto?  Unit must be in auto to respond to the commands.  The commands are not used if unit is in off or alarm.

klinquist, 2019-11-19:
@liltux   Bingo!  That explains it.    I only have propane hooked up so far, electrician is coming out tomorrow.

No way to turn it from auto to manual remotely? (This is a second home/airbnb  and I was hoping to keep it on manual - UNTIL the weekend where guests were arriving, then change it to auto).

(Or if I put it on auto - then tell it to turn off via genmon - would that essentially be in manual?)

klinquist, 2019-11-19:
@liltux  ... Just put it in auto, generator started up... and now only the "stop" button in genmon was highlighted - but it did not stop the generator.

I suppose I'll wait until I get the electrical hooked up and re-test.

liltux, 2019-11-19:
@klinquist when you send the stop command the unit may trigger cool down and will shutdown after 60 seconds instead of immediately.

klinquist, 2019-11-19:
@liltux Would I see that on the control panel/display? (I didn't see any change on the display)

jgyates, 2019-11-19:
The display should say "Cooling Down" if you are in cooling down mode.

klinquist, 2019-11-20:
Now that the generator is on line power and has no faults, I can remotely start/stop it.

Now another issue.. When the generator boots, it continually looks for wifi.  I have to plug in the wifi controller, wait for it to find wifi, then unplug the wifi and plug in the genpi.

Can I somehow tell my generator not to ever look for wifi?

liltux, 2019-11-21:
@klinquist when you get on the controller navigate to the edit menu.  Under the edit menu you can select to "redo" wifi.  This will turn off the wifi feature of the controller.

klinquist, 2019-11-21:
Thank you @liltux.  One more quick question-
Do you know of any way to switch the generator from “off” to “auto” remotely (and from auto back to off)?  This was my #1 reason for getting the GenPi, but it doesn’t seem to be possible.

To make it clear, I want to switch modes while I am on utility power.

The house is an airbnb, and I’d like to keep it ‘off’ most of the time (so it doesn’t switch on while nobody is here), then switch it to “auto” when I am expecting a guest.

jgyates, 2019-11-21:
@klinquist 

the  controller does not support this functionality over the modbus  interface for Evolution Air Cooled units. This feature is supported only on Evolution Liquid Cooled controllers. This is a limitation of the Generac firmware.

jgyates, 2019-11-21:
I am closing this issue since your original problem is resolved. Feel free to post related questions to this thread.

closed 2019-11-21 by jgyates