jgyates/genmon #328: Updates not working

UPS99, 2019-11-21:
Use the template below if you have an issue or want to report a bug. If you have a question or a feature request you can ignore the questions below.

***NOTE:*** If you are having issues with your serial connection, please read this page before posting:

https://github.com/jgyates/genmon/wiki/3.6---Serial-Troubleshooting

If you are having other issues, please see the following page:

https://github.com/jgyates/genmon/wiki/3.5---General-Troubleshooting

If you need to send you logs and registers to the developer, if you email is setup and working properly you can click send your logs on the About page in the web interface.

### Expected Behavior

Current version v 1.13.29.  Tried to upgrade to latest version v 1.13.32 

### Actual Behavior

Displays "Downloading Latest Version" and seems to download OK, then displays "Restarting..." for 10 seconds or so, but then there is a brief flash and it returns to the screen which still shows "You are not running the latest version".  Tried in IE 11, Chrome 78, Firefox, and PaleMoon on 3 different PC's - but the results are the same.  Rebooted the PI manually and tried again - same problem.  This worked OK  for last update to v 1.13.29.  No changes made since then.  Everything else works normally.

### Steps to Reproduce (including precondition)

### Screenshot or Pictures relating to the problem (if possible)

{Please write here}

### Your Environment

- Generator Model: 6244
- Generator Registers: Submit via the About page
- Genmon Version: 1.13.29

jgyates, 2019-11-21:
There is a know issues with updating with Firefox or Palemoon:

https://github.com/jgyates/genmon/wiki/Appendix-D-Known-Issues

I have validated that the upgrade still works as before with Chrome, IE and Edge. I would suggest trying to perform the upgrade from the console as the problem is probably related to the pi failing to perform the update.

1) log into the pi via ssh
2) change folders to the genmon directory:
     
        cd genmon

3) perform the upgrade from the command line:

       bash ./genmonmaint.sh -u

If you see errors on the last step then you can post them here.

UPS99, 2019-11-21:
I had been using IE for the last several updates and it worked fine.  I only tried the other browsers after IE failed this time.  I will try the manual update a bit later and see what happens.  

 

https://github.com/jgyates/genmon/wiki/Appendix-D-Known-Issues

3.   cd genmon

4.      perform the upgrade from the command line:

5.  bash ./genmonmaint.sh -u

jgyates, 2019-11-21:
You could have a corruption in  your SD card that is preventing the update.  The  manual process will show this, if this is the case.

dennis48755, 2019-11-21:
FYI, I've been using Firefox (64bit) without issue.

Dennis

jgyates, 2019-11-21:
I have duplicated the  issue with both 32 and 64 bit versions of Firefox. Note that this is only an issue when you upgrade via the web interface. All other functionality works as expected.

UPS99, 2019-11-21:
I had brought up the issue of PaleMoon and Firefox not working properly for upgrades several months ago and had only used IE 11 since then.  However even then, the update would succeed but with many alert messages.  The issue now is different.  No errors, delays, or email alerts.  The Restarting message just disappears and the same web page reloads immediately.  

 

I am remote from it right now and can access the Genmon web page but don’t have SSH access – I will be back there in about an hour and will try the manual reboot.

UPS99, 2019-11-22:
Running the update manually immediately showed the problem was that the PI could not reach https://github.com/jgyates/genmon/... or anything on the web.  

There are 2 routers involved here – one for external internet access – and clearing the arp table on that router fixed the problem.

This issue can be closed.

Thanks, Jason

 

https://github.com/jgyates/genmon/wiki/Appendix-D-Known-Issues

jgyates, 2019-11-22:
I am glad it is working and thanks for letting us know what the fix was.

closed 2019-11-22 by jgyates