jgyates/genmon #329: Feature or Fork Request - X86_AMD64 method?

volkswagner, 2019-11-24:
I'm curious if I can adapt this to a 64bit install (non-ARM, non-Pi).

The reason is I'd like to combine the Genmon with other services, such as
TP-Link's Omada controller, which I assume will not work on ARM architecture.
https://www.tp-link.com/us/support/download/eap-controller/#Controller_Software

Pardon my lack of knowledge on this request. I understand the Pi is inexpensive, and
if this is a big task, I'm confident just adding the Pi to my environment will be the path
of least resistance.

Please let m know if it is trivial or a major rewrite to get Genmon on 64bit Debian on Intel CPU.
My use case is to use the following hardware:
https://www.amazon.com/gp/product/B07JKW5T7R/ref=ppx_yo_dt_b_asin_title_o02_s00?ie=UTF8&psc=1

Thanks for your consideration.

liltux, 2019-11-24:
Genmon is based on python, javacript, css...  I do not believe it is rpi specific.  I have not tried but if your OS is at least debian based and linux, the commands should all work.

jgyates, 2019-11-24:
@liltux is correct. There is nothing specific in the code about the pi. I know that at least one person is running genmon in a virtual machine running Ubuntu (using serial over IP). 
this page has the base requirements  for the project:
https://github.com/jgyates/genmon/wiki/Appendix-E---Running-on-Non-Linux-Platforms

This thread has info on running on non PI platforms.

https://github.com/jgyates/genmon/issues/40

volkswagner, 2019-11-24:
Thank you @liltux and @jgyates for the information and quick reply!

This is good news. I will check out the links provided. 

Cheers!

jgyates, 2019-11-25:
I am closing this thread but if you have questions related you can post to this thread.

closed 2019-11-25 by jgyates