jgyates/genmon #330: Is there any way to tell the *next* exercise time?

klinquist, 2019-11-24:
I have my generator set to biweekly, I am wondering if there is a way to tell whether this is an “on week” or “off week.”

(brand new generator that has never exercised itself)

A great feature would be for genmon to offer an iCal link that contained the recurring events :)

jgyates, 2019-11-25:
Probably the easiest way is to wait for the first exercise, then look at your run log to see the last exercise time.

If memory serves once you set the exercise time for biweekly, the next weekly occurance is skipped. For example if you set it up on a Tuesday for a biweekly exercise on Monday, it would skip the next Monday and exercise on the 2nd Monday.

klinquist, 2019-11-25:
Thanks!   Might dig into the python to see if I can show the next exercise :).  Will be sure to submit a PR if I do.

closed 2019-11-25 by klinquist

jgyates, 2019-11-25:
The issue that makes this problematic to implement is the controller does not export the next exercise time / date. If genmon tries to keep up with the next exercise time is can not account for an exercise time that was setup before genmon was loaded.

klinquist, 2019-11-25:
Understood - genmon would have to wait until an exercise completed while it was loaded until showing the next..