jgyates/genmon #331: Incorrect power output?

klinquist, 2019-11-27:
- Generator Model:G0062770

I have a 20kW generator and Genmon is showing 20kW for the power output.  My house certainly isn't drawing the full capacity of the generator.   Any thoughts on how to get a good load estimate?

Status :

    Engine :
        Switch State : Auto
        Engine State : Running
        Battery Voltage : 14.30 V
        RPM : 3602
        Frequency : 59.00 Hz
        Output Voltage : 238 V
        Output Current : 83.33 A
        Output Power (Single Phase) : 19.83 kW
        Active Rotor Poles (Calculated) : 2

    Line :
        Utility Voltage : 0 V
        Utility Voltage Max : 0 V
        Utility Voltage Min : 0 V
        Utility Threshold Voltage : 156 V

Maintenance :
    Model : G0062770
    Generator Serial Number : 3004828432
    Controller : Evolution 2.0, Air Cooled
    Nominal RPM : 3600
    Rated kW : 20 kW
    Nominal Frequency : 60 Hz
    Fuel Type : Propane
    Engine Displacement : 999 cc
    Ambient Temperature Sensor : 41 F

    Controller Settings :
        Calibrate Current 1 : 1501
        Calibrate Current 2 : 1504
        Calibrate Volts : 1039
        Nominal Line Voltage : 240 V
        Rated Max Power : 20 kW

    Exercise :
        Exercise Time : Biweekly Wednesday 12:00 Quiet Mode On

    Service :
        Service A Due : 198 hrs or 11/16/2021
        Service B Due : 398 hrs or 11/16/2023
        Battery Check Due : 11/16/2020
        Total Run Hours : 1
        Hardware Version : V1.00
        Firmware Version : V1.08

jgyates, 2019-11-27:
Number 5 on this page (https://github.com/jgyates/genmon/wiki/Appendix-D-Known-Issues) should help

klinquist, 2019-11-27:
Interesting... Doesn't seem that it's a "mutiplier" issue since it always shows 20kW out.  Any troubleshooting or additional info I can provide?

jgyates, 2019-11-27:
If you look on the registers page in the web interface (click the icon in the upper right once), then look at register 003a (CT Sensor EvoAC). The is the raw sensor value that the controller is giving the software. In observation this value appears to change with the load in most models (22kw EvoAC does not have current transformers so this does not apply to the 22kw model). The value read from this register is applied to the formula on the page link I provided above.

Regarding your comment that it does not seem to be a multiplier issues, the software will not allow the result of the calculated power to be greater than the rated power of the generator or lower than zero to be displayed, so I would look at register 003a to see if it changes. 

Since you have a Evo2.0 you should be able to view the amps the controller is calculating from the dealer menu on the LCD display. To access the dealer menu you can enter the following key sequence: Up, Up, ESC, DN, Up, ESC, Up, ENTER from the main display, then enter the dealer menu then select Display under the dealer menu. From there scroll until you see Output Amps 1 and Output Amps 2. If your unit does not have current transformers then you will not see these values change when under load. If you log the values of these two Amp readings and the value of the register 003a then it should be straightforward to determine the value of the divisor and offset. I would take a measurement of the Amp 1 and 2 and 003a when the generator is running with a low load and a high load.

Let me know if you have any other questions.

klinquist, 2019-11-27:
Thanks - I'll have to do this next time I'm at the property.   It's my second home/airbnb... the power is out right now.

CT Sensor (EvoAC)
--
(003a)
HEX: 00000a29
DEC: 2601 \| HI:LO: 0:0

[image]

klinquist, 2019-12-19:
I followed your instructions and in Dealer/Display, I did not see Output Amps 1 & Output Amps 2.  Maybe my unit does not have the transformer to perform this measurement?

I looked at 003a with both my electric hot water heater on and off.    No other high load appliances were running.  With the heater off I would guess that the house is using <20a.  I saw hex 924 / dec 2340.

When the heater on, I saw hex add / dec 2781.  The hot water heater uses about 20a (power meter on it shows about 4500w@220v).

Am I correct in thinking that I may need a clamp ammeter to measure my actual house usage to form a reliable multiplier for 003a?

Thanks!

jgyates, 2019-12-19:
Yes, you are correct. You would need to measure both 120V legs of your main supply (Assuming you are is north american with two 120V lines). Your transfer switch could possibly be a place where you could perform this measurement.

jgyates, 2020-01-23:
I am going to close this issue for now however feel free to post to this thread if you have any additional data or questions.

closed 2020-01-23 by jgyates