jgyates/genmon #332: MODBUS molex connector power question

lmamakos, 2019-11-27:
I have a question about the molex connector on the Generac generators.  I'm working on designing a small PCB that would plug into the GPIO connectors of the Raspberry Pi.  For me, this started from a desire to add some thermocouples to monitor some of the high temperature parts of the generator where the DS18B20 devices wouldn't be appropriate.  

I'm going to add an RS-232 to TTL level converter for the MODBUS connection that's going to be galvanically isolated from the Raspberry Pi and it's power supply, eliminating possible ground loops and other issues related to how the Raspberry Pi is powered.  I'm planning to power the "generator" side of this circuit from the +5V on the molex connector; it's probably only going to require around 50mA or so.   

Is this 5VDC power source generally available on that connector across the product line?  I'd rather like to avoid having to add in provision for another regulator from a 12V? (something else?) to 5V just to run a couple of chips.

I'm also trying to see if I can bring out some of the other GPIO signals to connectors or header pins for wiring up external alarm/status LEDs, not sure if that's going to fit or not.  If this all comes to pass, I'll post the PCB design somehwere for others to build.  However, it uses some tiny surface mount parts, and the thermocouple interface devices have a buried pad on the bottom of the package and will need to be reflow soldered with a hot air tool or in an oven..

jgyates, 2019-11-27:
While I do not now definitively, 5V does appear to be present across the Evolution and Nexus (residential) controllers.  For industrial controllers this would not apply as they do not have a molex connector. The 5V is available on Evolution 1.0 and 2.0, however Evolution 2.0 controllers also provide 12V  on the molex (https://github.com/jgyates/genmon/wiki/3.1--Making-a-Cable)

If this works out, I can give you access to post the PCB design to the repository or post them for you.

jgyates, 2019-11-29:
I am closing this issue for now, feel free to post to this thread if you have questions on this topic. Let me know how the PCB works out. It sounds interesting.

closed 2019-11-29 by jgyates