jgyates/genmon #333: Packet Timeouts started showing up

jimland58, 2019-12-05:
If you need to send you logs and registers to the developer, if you email is setup and working properly you can click send your logs on the About page in the web interface.

### Expected Behavior

No packet timeouts

### Actual Behavior

About a week ago started seeing a small percentage of packet timeouts but has been increasing.  I had been running 172 days.  Decided to upgrade to the latest version V1.13.32 last night.  Now that it rebooted starting see that actual percentage of packet timeouts.   Sent log files to developer.

   Serial Stats :
        Packet Count : M: 10163, S: 2360

        CRC Errors : 0

        CRC Percent Errors : 0.00%

        Packet Timeouts : 7802

        Packet Timeouts Percent Errors : 76.77%

        Modbus Exceptions : 0

        Validation Errors : 0

        Invalid Data : 0

        Discarded Bytes : 0

        Comm Restarts : 0

        Packets Per Second : 0.53

        Average Transaction Time : 0.0314 sec

### Steps to Reproduce (including precondition)

{Please write here}

### Screenshot or Pictures relating to the problem (if possible)

{Please write here}

### Your Environment

- Generator Model:     G0070432, 22kw, Evolution 2.0, air cooled
- Generator Registers: Submit via the About page
- Genmon Version: V1.13.32

jgyates, 2019-12-05:
I received your logs. Other than the timeout error that is logged there is nothing unusual going on. This leads me to thing that you have one of two possible problems.

1) You cable is marginal. Possibly the vibration of the generator has make the connection less than adequate. Examine the connections and try to adjust them.

2) You controller needs to be restarted. I have seen issues with serial communications that have been resolved by rebooting the controller. This page has info on how to reboot your controller:

https://github.com/jgyates/genmon/wiki/Appendix-I-Power-Cycle-the-Evolution-or-Nexus-Controller

jimland58, 2019-12-05:
Resetting the controller fixed it!  

Should have known to do that first.  Last time I had and issue with a complete loss of communication, resetting the controller fixed it.

jgyates, 2019-12-05:
I am glad it fixed the problem and thanks for the feedback.

closed 2019-12-05 by jgyates