jgyates/genmon #334: Factory WiFi connection with Evolution 2.0 controller

tcflint, 2019-12-05:
I have my Pi setup and running (huge thanks by the way), in doing this i needed to unplug the factory connector that I assume goes the WiFi because my crappy MobileL!nk app isn't showing connection.

Can the two coexist?  Is the any pros and cons of not having it hooked up?

I apologize if this has been covered but I didn't see it mentioned anywhere.

jgyates, 2019-12-05:
From my understanding yes they can. See this thread:  https://github.com/jgyates/genmon/issues/183

FYI, you can search the issues on this project by Labels. There is a label for "Evolution 2.0" so you can search on all issues with this label with this URL:  https://github.com/jgyates/genmon/issues?utf8=%E2%9C%93&q=label%3A%22Evolution+2.0%22+

You can also select the label to search via the web interface. (Note: be sure to search on issues that are open and closed by removing the "is open" attribute in the search menu.

jgyates, 2019-12-06:
I am closing this issue but feel free to comment in this thread if you have related questions.

closed 2019-12-06 by jgyates

jshep321, 2023-10-25:
@jgyates I propose adding a one line reference this to the readme.  Some (such as myself) will want both, especially during the warranty term.

jgyates, 2023-10-25:
done: https://github.com/jgyates/genmon/wiki/3.1--Making-a-Cable