jgyates/genmon #335: Novice using thonny for serial config testing...need help.

waynewarburton, 2019-12-06:
I'm a novice having troubles checking the serial port...more-so using the Thonny software. When I run "serialconfig.py", text reads "option not selected". I know it's looking for me to input "e, r, or c" to enable, restore, or check but can't figure out how to input this information in.

Hardware I am ok with, software...very new to it. The site is very geared towards experienced programmers and this I am not. I have everything wired on my desk and running Thonny (haven't worked with basic level software since the late 80's). I have read many links today and none touched on this (almost all I did today was read links other than blow out the driveway). Any help would be greatly appreciated!  It's something basic I am missing.

I have tried using console command 'sudo python serialconfig.py -e' and text comes back 'python: can't open file 'serialconfig.py': [Err02} No such file or directory.  I do have the program installed in /home/pi/genmon/otherapps

jgyates, 2019-12-06:
I am not familiar with Thonny. I use a console to run python programs. Once you open a console, from a console you type:

   sudo python serialconfig.py -c
the "sudo" part runs the program with administrator (administrator is named "root" on linux systems), the "python" is running the python program which interprets the program file. The python source file is "serialconfig.py" and the "-c" is the options passed to serialconfig.py.

Also, instead of posting questions to a random thread, please either post to a new thread or to a thread that is specific to the topic in question. Adhering to this helps keep the questions organized so others can benefit. From this link you should be able to create a new issue or search for old issues (closed issues will not show up unless you search for them). Issues are tagged by topic so you can search them easier.

You have to be in the directory that the serialconfig.py file is located. when you open a console you are typically in the directory of the logged in user (assume the user is pi since this is the default). so when you open a console you will likely be in the /home/pi directory. If you installed genmon per the instructions it will be located in the /home/pi/genmon directory. to get there from the /home/pi directory type:

     cd genmon

This page has more details on navigating directories. This is the same concept as folders on a Windows or Mac.

You are getting the "can't open" error because the file serialconfig.py is not in the /home/pi/genmon directory, it is located in the /home/pi/genmon/OtherApps directory. So you could type:

    cd genmon/OtherApps 

 from the /home/pi directory then you can type:

     sudo python serialconfig.py -c

jgyates, 2019-12-06:
forgot to include the link:

https://www.techrepublic.com/article/linux-directory-navigation-tips-and-tricks/

waynewarburton, 2019-12-06:
Now we're getting somewhere.  Old school DOS directory stuff...haven't used it in years but that worked.  Thanks for the links.  Problem was I was in the wrong directory as you suspected.  By running Thonny, it gives the impression to just type which of the 3 options you want.  Another lesson learned is Capitals are very important (OtherApps is not otherapps)!

Checked out the link. Opened the console and followed your directions and
was able to configure the serial port, reboot, then do the loopback test successfully :)

Using Thonny,  serialconfig.py doesn't let me enter a selection (e,...) when I hit the green 'Run current script' in for 'serialtest.py' it runs starts the test, looking like it doing the test, and then says error.   I don't think it works well for setting this up and will go old school from here on in!

ECM56, 2019-12-07:
You must stop the monitor program from running before running the serialtest.py because it will be using the serial port.  

 bash ./startgenmon.sh stop

This is described in the Genmon project wiki (See "Run a Loopback Test" on this page: https://github.com/jgyates/genmon/wiki/3.6---Serial-Troubleshooting).

ECM56, 2019-12-07:
By the way, you should not need to use Thonny - these scripts are ready to run using Python.  You do need to be in the directory containing the script - I made that mistake when I first tried to run it.

waynewarburton, 2019-12-09:
Still going under the Novice.  Just plugging along slowly and changed the config files for my set-up.  Getting ready to install the comm cable to the generator next .  Weather is hovering at or below freezing here so not in a hurry for that.  I just tried to run the ClientInterface without the gen hooked-up hoping to see the display board and getting "WARNING System Initializing.  Not receiving data. : ,"  Kind of expected something like this...This is looking promising! 

I am going to do a temporary set-up with it in the generator's enclosure for the commissioning to reduce my self-induced frustration variables and once that's proven and the weather improves...
considering going with a 12-14' CAT 5 LAN wire for the serial cable and another pair for the power from the battery (still need to buy some power wire) to allow mounting the new control box inside my garage or house for a stronger WIFI signal (one less brick wall to go through as I currently have an extender without a UPS battery back-up on it).  I'm hoping this isn't too long of a run for signal attenuation. 

I've appreciated your fast response and patience.  Noobies can be frustrating!  If I have any software questions that arise, would it be ok to PM you or do you prefer to keep it in this forum?

jgyates, 2019-12-09:
My preference is to keep it in the forum since this allows others to learn from the documentation trail.

ECM56, 2019-12-10:
I am only a step ahead of you.  I installed a box on a column near the generator and ran a watertight piece of flexible conduit between the generator and the box.  There was a one inch hole already cut in the generator housing with a cap on it so I was able to use that.  I was able to string my cable through the conduit but the cold and mud finally got the best of me.  Today it was cold and raining so I got nothing more done.

This shows the pre-cut hole with the chrome cap in it - that's where I will run the conduit.

[image]

Here you can see the flexible conduit installed to the pre-existing hole.

[image]

Here is the weatherproof box attached to the concrete column.  It's similar to the boxes used by the electrician who installed the generator.

[image]

The small white wire is CAT6 going back to a POE ethernet hub in the house.  I still need to finish tying that down.

waynewarburton, 2019-12-10:
Temperature dropping below zero.  I have the box ready for a temporary install and testing.
[image]

I haven't tried to communicate with it from my desk computer.  So far all testing was performed using the Pi.  More reading required.  Looks like I need to install Python on my desktop under some other drive than 'C'?  Then should be able to communicate through the IP address?

ECM56, 2019-12-10:
The GenMon program creates a web site that runs on your pi at port 8000.  That allows you to operate the program using just a web browser.  I also setup VNC server on the pi which allows me to use a VNC viewer to open a graphical terminal window from my Windows PC to the pi.  That allows me to do anything (I think) I could do with the pi connected to a monitor, mouse, and keyboard.

ECM56, 2019-12-10:
SUCCESS!  I kept running into problems when I put the pi in the box - I would lose the serial connection.  Finally I took it all back to my desk in my office and looked at it very closely.  I wiggled all the pins on the serial board and one was loose.  It had a cold solder joint.   I made that board up years ago and have used it on several projects but I must have just barely been holding on.  I touched it up with the soldering iron and re-installed it and it is now working.  I just started the generator from the web browser on my Windows PC - COOL!!

jgyates, 2019-12-10:
Excellent! I am glad it is working for you. Thanks for the donation. I am going to close this thread but feel free to post related questions in this thread if needed.

closed 2019-12-10 by jgyates