jgyates/genmon #336: Unable to Login

dave-sdc, 2019-12-16:
My manager was setting up the monitor for one of our generators.  While doing so he enabled an option that is preventing him from logging in.  He believes it was "use advanced login encryption".
How can I reset the software to clear this issue.
Thank-you

liltux, 2019-12-16:
@dave-sdc you will have to ssh into the pi and change the configuration file in /etc/genmon/genmon.conf manually.  Then you will be able to run 
sudo bash /home/pi/startgenmon.sh stop 
and then
 sudo bash /home/pi/startgenmon.sh start 
This will restart genmon software. Or reboot the pi
sudo reboot
And let everything start over.  You can review the genmon.conf file on here to verify which lines need to be commented out and remove the leading # on the correct line.  
If you put https:/[genmon pi address]
should pull up the correct page and ask for a user name and password that was setup by your manager.

jgyates, 2019-12-16:
My response assume you are familiar with ssh and editing a text file on linux. You can log into the pi via ssh. This will give you a console for the linux operating system. from there you can edit the file /etc/genmon/genmon.conf 

The setting for genmon.conf are documented here:

https://github.com/jgyates/genmon/wiki/1----Software-Overview#configuring-genmonpy

You are probably referring to the setting that enables encrypted web access. To disable this change 

        usehttps = True 

to 

     usehttps = False

in the genmon.conf file.

Alternatively you can copy the file in /home/pi/genmon/conf/genmon.conf to /etc/genmon/genmon.conf to reset all the settings to the default.

dave-sdc, 2019-12-16:
I'm not showing a genmon folder under /etc, the genmon.conf ( as well as other genmon files) are in the root of /etc.  When I try to edit the file I receive "no write permission" error.  I viewed the file, and it shows usehttps is set to false.

jgyates, 2019-12-16:
you are likely running an older version if there is no /etc/genmon directory. You can edit the file in the /etc/ directory. You can use the sudo command to gain administrator privileges. Assuming you are using the nano text editor you can type:

     sudo nano /etc/genmon.conf 

this will allow you to edit the file. The usehttps setting was a guess based on your "use advanced login encryption". There is not setting that matches that description but usehttps is the closest. Without knowing what is happening it is difficult to give you any guidance. You said it prevented him from logging in, does the web interface pull up and the password is incorrect, or is something else the problem. I also don't know the URL you are using so your port number could be off. You can compare the files /etc/genmon.conf with the genmon.conf file in /home/pi/genmon/genmon.conf (or /home/pi/genmon/conf/genmon.conf, depending on the version you are using) to compare the changes that were made from the default values. This may be the best way to determine what changed.

     comp /etc/genmon.conf /home/pi/genmon.conf 

This should display any differences between the two files which will allow you to determine what was changed.

dave-sdc, 2019-12-16:
Yes, it was the not loading the web interface.  Comp command not found when I attempted to compare them.  I manually compared them, and while doing so found the setting for HTTP port.  Using the port I was able to load the web interface without issue.  I will verify that that was the issue he was having and update as soon as I can.  Thank you for your time.

jgyates, 2019-12-16:
use the command diff instead of comp. Sorry, comp is an old DOS command, diff is the linux variant.

dave-sdc, 2019-12-16:
That's ok.  Has been 25 years since I took linux/unix training .  File was easy enuff to check manually.  :)

jgyates, 2019-12-19:
I am going to close this issue. If you have related questions feel free to post them to this thread.

closed 2019-12-19 by jgyates