jgyates/genmon #338: Attempted to make a static IP address and config file corrupted

waynewarburton, 2019-12-20:
Well it was working very nicely...my IP address on my LAN changed and I lost the ability to use ClientInterface on my desktop computer.  I went into my router and found it went from 10.10.10.231 to 10.10.10.229.  I went to the new address and everything worked well again.  did some reading and decided to get a static address for the Pi so went to my TP-Link router and set it to Raspberrypi  C6:71:54:D4:F6:31  10.10.10.229 (I used the same number...I think should have set it to an unused number).  Also played at the Pi and was looking at /etc/dhcpd.conf.  Somewhere along the way, I lost my monitor that is plugged into the Pi so I rebooted the Pi.  Monitor is back up and running but can't run the program now getting a configuration error.   

How do I run my backup config file genmon_backup.tar.gz?  Currently have the back-up file on my desktop computer.  Will this fix my problem?  Yup, I'm a newbie. I'd hate to start all over and don't think it will be required...

jgyates, 2019-12-20:
What is the configuration error? Is this an error from the pi or from genmon? The answer to this will determine how to fix your problem.

You can copy the genmon_backup.tar.gz file to the pi and decompress the file with this command:

     tar -xzf genmon_backup.tar.gz

The command need to be executed in the same directory as the file. This will put the config files in a directory and then they can be copied to the /etc/genmon directory. Once this is complete then genmon can be restarted.

If other non-genmon files are corrupt, then I would suggest reinstalling the operating system.

waynewarburton, 2019-12-20:
More detail...
when I try to run genmon using
 "./startgenmon.sh start", starting genmon python scripts
Error: E
mpty configuration found
Error validating config. exiting

jgyates, 2019-12-20:
I am assuming you are using relatively recent version of the software (i.e. at least V1.13.11 - 2019-05-20). If not then the following does not apply.

This message occurs when the contents of the genloader.conf file is not found. A simple solution would be to copy the file again. From the /home/pi directory type:

      sudo cp ./genmon/conf/genloader.conf /etc/genmon

Then from the genmon directory you can type ./startgenmon.sh restart

waynewarburton, 2019-12-20:
I do believe I have the most current version.  I'll try this right now.  Back in a bit.

waynewarburton, 2019-12-20:
You are a GOD!  Worked perfectly.  Any words
 of wisdom for setting the static IP before I screw this up again :)

waynewarburton, 2019-12-20:
OK.  I changed to a static IP on my LAN with a different number.  Rebooted the Pi and she went straight to that IP address.  Nice!
I think the only outstanding thing I want still is to automatically run genmon on boot-up.  I saw the instructions are on the site somewhere just haven't got around to it.  Perhaps tomorrow's mission.  Powered from the generator's battery so should only be a problem when disconnecting the battery for maintenance.

jgyates, 2019-12-20:
I would set the MAC address (C6:71:54:D4:F6:31) to the IP in your router like you did before (static IP DHCP assignment). If you do this then the pi will continue to operate in DHCP mode but your router will always assign the same IP to that specifici raspberry pi. If make the IP address 10.10.10.229  you may have to reboot what ever was using 10.10.10.229 . I use this method, but my router hands out DHCP IP address from 100 to 254 and I reserve IP addresses 2 to 99 to static IP addresses. This avoids the situation of having to reboot other network devices if I add a static IP.

jgyates, 2019-12-20:
I am going to close this issue for now but feel free to post in this thread if you have any follow up data or questions.

closed 2019-12-20 by jgyates

waynewarburton, 2019-12-20:
You have been very knowledgeable and helpful.  One thing I did notice, when running the generator without a load, the power meter on ClientInterface was indicating 21KW with no load.

jgyates, 2019-12-20:
Hi @waynewarburton 

You may need to calibrate the way that the software calculates the current. This page has more info (see # 5 on this page):

https://github.com/jgyates/genmon/wiki/Appendix-D-Known-Issues

Also there are several closed issues that deal with this topic to give you some background:

https://github.com/jgyates/genmon/issues?utf8=%E2%9C%93&q=label%3APowerDisplay+

Let me know if you have any other questions.

ECM56, 2019-12-22:
@waynewarburton  - I'm sorry I missed your posting.  I was struggling with another project, the PiHole ad blocker, and the system (Ubuntu 18.04) I was running that on did almost the same thing to me.  I thought I had configured the IP address to be static but Ubuntu has this thing called Network Manager that likes to change the network configuration.  Once I learned how to disable that, life was good again.  It sounds like you have GenMon working.  It is very cool.  My generator is running the weekly exercise right now.  Take care!

waynewarburton, 2019-12-24:
No problem.  All is good and running.  Very nice system.  I have a LAN cable run for the future.
I've developed a habit of watching a video before doing things I have a pretty good idea of how to already...this project is a niche segment of the market.  Steep learning curve on this one!
I've attached my install pic's from a Generac 20KW unit (with Honeywell decals).

[image]
[image]
[image]

waynewarburton, 2019-12-24:
Just one piece of info was missing on the write-up and it may help others later in life.
I did encounter problems receiving emails through Yahoo.  After a lot of reading and verifying, Yahoo Mail was automatically blocking the genmon emails.  I needed to change a setting to 'accept from less secure sources' (something like that).

 I never did set-up my text but it looks like some of the providers accept texts from emails and fire them out.  I didn't bother looking into it very far as when we are travelling, we go in airplane mode and don't receive texts.  Emails from web based servers are always available.

waynewarburton, 2019-12-24:
Sorry, it was the Gmail account that was rejecting the emails.  I added another account later when it was working.  That was also a nice part of the program adding more email addresses.
Anyways, have a Merry Christmas and Paypal has a gift waiting for you!

jgyates, 2019-12-24:
Thanks Wayne, Merry Christmas

ECM56, 2019-12-24:
Looks great!  

Happy holidays to you both!

waynewarburton, 2019-12-25:
Thanks.  Forgot to mention...the run for the power (from battery) and communications was approximately 18' of #12 gauge wire to get it into the basement with no issues in voltage drop or significant signal loss.  Info may come in handy for you some day.  I do like your idea of the PO and wish I incorporated that in the design stages.   I did have the extra #12 wire at home so that was free ;)

waynewarburton, 2019-12-25:
PO-= POE..

ECM56, 2019-12-25:
Thanks!  That is good information.  I am a little worried about the Raspberry Pi being outside, even though it is housed in a watertight housing.  If I have any issues with that, I may move it indoors.  My run might be a little further but not by much.  I would feel more comfortable with it indoors.

buzzshot, 2019-12-27:
One other option that also works is running 20ga 4 conductor shielded. 2 wires for the modbus and just need one wire for a link from the com on the generator to the com on the pi. Then there’s no floating voltages I am running about 70’ cable so far no issues into my house

ECM56, 2019-12-27:
Perfect.  Thank you.  I have my pi connected to the generator with 3 wires (T, R, and added ground) since I am using POE to power the pi.  I am definitely going to consider this idea . . .

waynewarburton, 2020-01-02:
Just wanted to let you know I have the Genmon sending texts to my phone via email in the notifications.  My cell service is with Koodo and I'm just east of Toronto.   I had a techy Bell friend was over and I was showing him your programming. He said the following should work and it does!  Just added my phone number "905xxxxxxx@msg.koodomobile.com" in the GUI Notification Recipients.  No subscription to Twilio required :)

zekyl314, 2020-01-02:
> Just wanted to let you know I have the Genmon sending texts to my phone via email in the notifications. My cell service is with Koodo and I'm just east of Toronto. I had a techy Bell friend was over and I was showing him your programming. He said the following should work and it does! Just added my phone number "[905xxxxxxx@msg.koodomobile.com](mailto:905xxxxxxx@msg.koodomobile.com)" in the GUI Notification Recipients. No subscription to Twilio required :)

Hey there, I have AT&T as call service and I have used their email to SMS gateway in the past.  I had intermittent success with it, so keep an eye out  for any issues you may see.  Calling into their support was pretty much useless as no one really knew what it was (But AT&T is huge a hard to deal with).  A few times I could get a case open and it would start working again in a few days.   

I am glad you found this free solution through your carrier as I loved using it before too!

ECM56, 2020-01-03:
I was able to setup the e-mail notification to send using my Gmail account and an "application password" specifically generated by Google to be used by GenMon.  It works great.  It sends an e-mail when the status of the generator changes.