jgyates/genmon #339: Unable to get weather

Appfrontier, 2020-01-07:
Generated a new OpenWeatherMap.org API key on the free plan. When starting GenMon I do not see any weather.....

I get the following error in the genome.log

2020-01-07 01:12:00,461 : Error in GetObservation: (5161911) : Your API subscri$
Reason: Invalid API Key provided : myweather.py:106

I have double and triple checked and I even ensured I used a city code that was found in their database.

Any ideas what is going on? If it is important I signed up about 1 hour ago

Appfrontier, 2020-01-07:
Seems I get weather but still seeing the errors in the log...Guess there are other supported features that I must not be able to get?

closed 2020-01-07 by Appfrontier

jgyates, 2020-01-07:
I have never seen an error message from the weather api related to a subscription. I would need to see the full message.  In normal operation you should not see any errors. If there are any issues with your internet / wireless connection or the open weather servers then you could see an error like this:

2019-12-04 13:00:04,237 : Error in GetObservation: (your location) : Exception in calling OWM web API.
Reason: API call timeouted
Caused by: None : myweather.py:108

On average I get about one error like this per month , for example when I reboot my wifi AP or router.

Appfrontier, 2020-01-07:
Ok. I copied the entire line so where would I get additional error details? It looks like those errors have stopped this morning so maybe it was a delay in provisioning the API key on open weather side.

jgyates, 2020-01-07:
that looks correct

pridesleap, 2020-08-04:
I realize this is a closed issue but I wanted to add something in case someone else should come upon this.

OpenWeather states that **it can take as much as 2 hours** to fully activate a new key. I ran into this problem this morning.