jgyates/genmon #340: Unable to get serial0 to ttyAM0

Appfrontier, 2020-01-07:
I setup the pi using the lite version. Ran the Venmo setup. got the gen mon web server up. Getting ready to set up serial cables and ran the tests.

I followed the instructions on the serial port setup again just to make sure.

No matter what I do I get the following when checking the ports

lrwxrwxrwx 1 root root 5 Jan  7 02:24 /dev/serial0 -> ttyS0
lrwxrwxrwx 1 root root 7 Jan  7 02:24 /dev/serial1 -> ttyAMA0

I checked the boot command line text and it has console set to tty1. none of the other test you had listed appears in the file. The config.txt appears fine as well.

I am using a Pi3b+.

Should I change the port for communications to serial1 or is there another item I can try to get serial0 to the correct setting. If I have to change to serial1 then what pins to I use for the config test.

SerialConfig.py -c shows

Checking : BT service disabled: OK
Checking : Serial console command line removed: OK
Checking : Is enable UART in boot config: OK
Checking : Is BT overlay disabled: OK
Checking : Serial console service disabled: OK

jgyates, 2020-01-07:
If you have rebooted your pi and you still see that ttyAMA0 is mapped to /dev/serial1, then the simplest solution is to just use /dev/serial1 with genmon. You can make this change via the web interface on the settings page (Port for Serial Communication setting).

Appfrontier, 2020-01-07:
> If you have rebooted your pi and you still see that ttyAMA0 is mapped to /dev/serial1, then the simplest solution is to just use /dev/serial1 with genmon. You can make this change via the web interface on the settings page (Port for Serial Communication setting).

Thanks for confirmation. 

I changed it to `/dev/ttyAMA0` is that fine or must it be `/dev/serial1`

Also, just making sure that all the pinouts, loopback tests, etc remain unchanged with the change in the serial port on the settings page.

jgyates, 2020-01-07:
Using /dev/ttyAMA0 should be fine.
 
pinouts are the same. For the loopback test  you just need to specify /dev/ttyAMA0 on the command line

https://github.com/jgyates/genmon/wiki/1----Software-Overview#otherappsserialtestpy-optional

Appfrontier, 2020-01-07:
Thank you

As an aside, just wanted to say thank you for your work on this project. Generac EOL my 2 year old Generac branded monitor and it never worked well anyway. Now they will not reactivate it and wanted me to buy a new one. This project has been fun and although Wifi it is already 100x better in features (should they all work once connected)

Appfrontier, 2020-01-09:
Reopening as the loopback tests fail when using /dev/ttyAMA0

Loopback tests do work when not specifying the parameter but then I believe it is using serial0 which is set to ttyS0

I will be connecting to the gen today but curious what the ramifications are with a loopback test that works with the serial0 ttyS0 but not the ttyAMA0 and if it needs to be working with the ttyAMA0 then is there a different pin assignment or configuration that needs to be done?

Everything I have tried to get serial0 back to ttyAMA0 has not appeared to work. The way I am checking is by issuing the command ls `-l /dev/serial*` which displays serial0 as ttyS0

liltux, 2020-01-09:
https://www.raspberrypi.org/documentation/configuration/uart.md

Appfrontier, 2020-01-09:
> https://www.raspberrypi.org/documentation/configuration/uart.md

I have done all of that and serial0 is still at ttyS0

liltux, 2020-01-09:
can you attach copies of your config.txt and cmdline.txt from your /boot folder?

jgyates, 2020-01-09:
"Reopening as the loopback tests fail when using /dev/ttyAMA0"

How does it fail? Can't open the port? timeout waiting for data? 

The genmon software must not be running when you run the loopback test. To top genmon if it is running:

     ./startgenmon.sh stop

Appfrontier, 2020-01-09:
> "Reopening as the loopback tests fail when using /dev/ttyAMA0"
> 
> How does it fail? Can't open the port? timeout waiting for data?
> 
> The genmon software must not be running when you run the loopback test. To top genmon if it is running:
> 
> ```
>  ./startgenmon.sh stop 
> ```

It fails with timeout waiting for data

genmon was stopped

jgyates, 2020-01-09:
And you do have Tx and Rx connected when you run the loopback?

Appfrontier, 2020-01-09:
> And you do have Tx and Rx connected when you run the loopback?

Yes. If I leave off the ttyAMS0 parameter the loopback passes

jgyates, 2020-01-09:
If you disconnect Tx and Rx (i.e. no looped-back) and run it leaving off the parameter, does it fail?

Appfrontier, 2020-01-09:
> can you attach copies of your config.txt and cmdline.txt from your /boot folder?

Config.txt

`# For more options and information see
# http://rpf.io/configtxt
# Some settings may impact device functionality. See link above for details

# uncomment if you get no picture on HDMI for a default "safe" mode
#hdmi_safe=1

# uncomment this if your display has a black border of unused pixels visible
# and your display can output without overscan
#disable_overscan=1

# uncomment the following to adjust overscan. Use positive numbers if console
# goes off screen, and negative if there is too much border
#overscan_left=16
#overscan_right=16
#overscan_top=16
#overscan_bottom=16

# uncomment to force a console size. By default it will be display's size minus
# overscan.
#framebuffer_width=1280
#framebuffer_height=720

# uncomment if hdmi display is not detected and composite is being output
#hdmi_force_hotplug=1

# uncomment to force a specific HDMI mode (this will force VGA)
#hdmi_group=1
#hdmi_mode=1

# uncomment to force a HDMI mode rather than DVI. This can make audio work in
# DMT (computer monitor) modes
#hdmi_drive=2

# uncomment to increase signal to HDMI, if you have interference, blanking, or
# no display
#config_hdmi_boost=4

# uncomment for composite PAL
#sdtv_mode=2

#uncomment to overclock the arm. 700 MHz is the default.
#arm_freq=800

# Uncomment some or all of these to enable the optional hardware interfaces
#dtparam=i2c_arm=on
#dtparam=i2s=on
#dtparam=spi=on

# Uncomment this to enable infrared communication.
#dtoverlay=gpio-ir,gpio_pin=17
#dtoverlay=gpio-ir-tx,gpio_pin=18

# Additional overlays and parameters are documented /boot/overlays/README

# Enable audio (loads snd_bcm2835)
dtparam=audio=on

[pi4]
# Enable DRM VC4 V3D driver on top of the dispmanx display stack
dtoverlay=pi3-disable-bt
max_framebuffers=2

[all]
#dtoverlay=vc4-fkms-v3d
enable_uart=1`

cmdlin.txt

` console=tty1 root=PARTUUID=6c586e13-02 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait`

Appfrontier, 2020-01-09:
> If you disconnect Tx and Rx (i.e. no looped-back) and run it leaving off the parameter, does it fail?

Yes it does

liltux, 2020-01-09:
Are you running the latest raspbian "buster"?

Appfrontier, 2020-01-09:
> Are you running the latest raspbian "buster"?

The lite version yes. Had to use that AFAIK as I configured everything via terminal and network with no UI or IO inputs

liltux, 2020-01-09:
try
dtoverlay=disable-bt
instead of
dtoverlay=pi3-disable-bt

supposed to compatible, but never know for sure.

Appfrontier, 2020-01-09:
> dtoverlay=disable-bt

No dice. Made the change then did a reboot

[image]

liltux, 2020-01-09:
can you also show the output of:
ls -al /dev/ | grep serial

Appfrontier, 2020-01-09:
> ls -al /dev/ | grep serial

[image]

liltux, 2020-01-09:
loop back passes at the rpi pins for ttl (tx and rx) with no options and fails if specify /dev/serial1?

jgyates, 2020-01-09:
As referenced in the link above it looks like your pi is setup to use the internal mini UART instead of the internal PL011 UART. The loopback tests appear to confirm this. I would suggest trying this on your generator and see if it works. I would guess that it does since the data rate is only 9600 baud and no flow control is used. If you get errors your best bet may be to reinstall raspbian. I would assume that something may of caused the os to change the assignments, possibly the order that things were done on start up, but that is just a guess.

Appfrontier, 2020-01-09:
> loop back passes at the rpi pins for ttl (tx and rx) with no options and fails if specify /dev/serial1?

Passes when using `serialtest.py` with pins 8 and 10. Fails when run-in `serialtest.py /dev/ttyAMA0`

So yes, passes on serial0 fails on serial1

I just connected to gen and with the port for serial communication set to `/dev/serial0` it seems to be working. I get data, am able to remote start, etc. Have not found the fuel consumption yet and gen was ran for about 10 mins but seems to be working.

Side not, I tried using the power from the Evo controller and got none. No lights at all on the RS232 or the PI. Went the battery route and everything powered up

Appfrontier, 2020-01-09:
> As referenced in the link above it looks like your pi is setup to use the internal mini UART instead of the internal PL011 UART. The loopback tests appear to confirm this. I would suggest trying [...]

Here is exactly how I did thing:

1. Made image on SD of raspbin lite
2. mounted the img on my mach to do a `touch ssh` command
3. put the SD with the image into the Pi, connected to Ethernet and powered up
4. ran `sudo raspi-config` to set the wifi settings and change the host name
5. `sudo reboot` -> removed LAN connection
6. Followed instructions here: https://github.com/jgyates/genmon/wiki/3.3--Setup-genmon-software

Not sure why I cannot get serial0 to be `ttyAMA0` but it appears to be working except for unable to power at all from molex connection so using battery, and I do not see NG fuel consumption anywhere but otherwise appears to be running fine despite the discrepancies

Appfrontier, 2020-01-09:
> If you did try it again, I would attempt it without mounting the image and touch ssh command. That is the only thing that is different than the way I do it. I typically make my first book with a monitor connected then enable ssh via raspi-config, the I do everything else headless.

If I ever have to do it again I will try that. Would have to get a USB keyboard and mouse to do so but they are cheap enough.

jgyates, 2020-01-09:
I will try your way also to see if I can duplicate. I was meaning to order a pi4 once the firmware resolved some of the power issues. Now is probably a good time to do that.

liltux, 2020-01-09:
Using the official raspberry pi documentation, using Windows, put an extensionless file in the boot partition ssh and copy in a wpa_supplicant.conf file with my network settings into the boot partition as well.  Thus has worked great for me headless.

Appfrontier, 2020-01-09:
> I will try your way also to see if I can duplicate. I was meaning to order a pi4 once the firmware resolved some of the power issues. Now is probably a good time to do that.

Just to be succinct I am using a Pi3B

Appfrontier, 2020-01-09:
> Using the official raspberry pi documentation, using Windows, put an extensionless file in the boot partition ssh and copy in a wpa_supplicant.conf file with my network settings into the boot partition as well. Thus has worked great for me headless.

On difference is I am using mac and I used the raspi-config to configure the wireless. Maybe by launching the config program it did something to the serial post stuff automatically....hmm

jgyates, 2020-01-09:
I have a pi3b also

jgyates, 2020-01-11:
Here is a data point:

I installed with the "touch ssh" method to boot with ssh enabled (i.e. allow pi headless boot). I am using a Mac also (mounted the img, cd to boot volume of the img, then type "touch ssh". After the initial boot this was the serial setup:

[image]

I performed an software update (sudo apt-get up, sudo apt-get dist-upgrade), then I changed the hostname, then installed git and put the genmon repo on the system, then I ran serailconfig.py. After a reboot I got this:

[image]

This is with a raspberry pi 4, so we are not 100% apples to apples. I will try with a pi3b when I have time, but this is the same behavior that I see on the other pi3bs that I have (except I did not do the "touch ssh" trick then).

joedr, 2020-01-23:
I am running the same issue with  pi3b+.  I just cannot get serial0->ttyAMA0 with may attempts.  The difference on my case is that I cannot get the loopback to work.  I tried a USB to RS232 and the script for loopback works using: python serialtest.py /dev/ttyUSB0.

Here is the config:
`dtparam=audio=on

[pi4]
# Enable DRM VC4 V3D driver on top of the dispmanx display stack
dtoverlay=pi3-disable-bt
#dtoverlay=pi3-miniuart-bt
#dtoverlay=disable-bt
max_framebuffers=2

[all]
#dtoverlay=vc4-fkms-v3d
enable_uart=1
`

Removed the "console=serial0,115200" from commandline.

liltux, 2020-01-23:
@joedr Are you using a USB to serial adapter or a ttl to serial adapter?

joedr, 2020-01-23:
I used only the USB to serial to check how the script worked and to ensure I was able to see what it was expected from the script.  After I was successful testing the USB, I moved to the GPIO ones following the wiki and could not get it to work.

Here is the permanent setup. (See attached picture)
[image]

joedr, 2020-01-23:
The main problem I am running into is that I cannot get the loopback to work on Pins 8 & 10 from the GPIO.  I removed all the wiring/adapter/RS232toTTL/etc until I get the loopback to work straight from the board as it does not make sense to test further (the RS232-to-TTL, DB9 Adapter and EPS MOLEX Ribbon).  Hope this makes sense.

jgyates, 2020-01-23:
I am assuming you have went through this list:

https://github.com/jgyates/genmon/wiki/3.6---Serial-Troubleshooting

If you have pin 8 and 10 connected can you run this test:

     cd OtherApps 
      sudo python serialtest.py /dev/serial1 

What is the output of this command:?

      ls -la /dev/ser*

https://github.com/jgyates/genmon/wiki/1----Software-Overview#otherappsserialtestpy-optional

joedr, 2020-01-23:
[image]
[image]
[image]

Appfrontier, 2020-01-23:
FYI: My serials look just like yours and the loopback only worked on serial0 as well despite it showing ttyS0.

I simply ignored it since it worked on serial0 and it has been connected to the gen and working fine....

While it does not solve the issue with serial0 not being ttyAMA0 hook it up to your gen and see if it is functioning. I was not able to find a way to get serial0 to be ttyAMA0

jgyates, 2020-01-23:
To validate what @Appfrontier is saying you can loop pin 8 to 10 and run the loopback test on serial0

What @Appfrontier and apparently you are experiencing is this: There are 2 serial ports in the pi system on a chip (SoC). The two internal serial ports are ttyAMA0 and ttyS0. In most cases ttyAMA0 is routed to the GPIO header, but for some reason ttyS0 is routed to the header in your case. Since the OS abstracts the underlying port mechanics with /dev/serial0 and /dev/serial1 names it really does not matter how the SoC is using its internal ports. It would be nice if this were documented somewhere as to why some times ttyAMA0 is used and sometimes ttyS0 is used, but as long as /dev/serial0 is working on a loopback on the GPIO pins 8 and 10 you should be good to go.

jgyates, 2020-02-01:
I am closing this issue as this does not appear to prevent genmon from operating. To me this appears to be some anomalie of raspbian and how it assigns serial ports to the underlying hardware. Feel free to comment in this thread if you have any additional data or questions.

closed 2020-02-01 by jgyates