jgyates/genmon #341: Add NG Usage Gauge or line on status page. Maybe (3) broken down into Outage vs non outage and total

Appfrontier, 2020-01-09:
Based on the post here: https://github.com/jgyates/genmon/issues/302#issuecomment-513499326

I can use the `cmd/power_log_json?power_log_json=0,fuel` to get a result

I have not had an outage since I installed the genmon but I did manually run the gen. When I use that command I get 

**3.35 cubic feet**

The problem is I cannot find this anywhere in genmon.

Reading this thread: https://github.com/jgyates/genmon/issues/293

it appears the usage is only shown for outages.

IMHO it would be useful to know how much NG is being consumed or has been consumed since reset.

Ideally for those of us with NG there would be three categories:

1. Exercise or non-outage (manual)
2. Outage
3. Combined usage

Having a graph like the kW output or similar over a period of time would be great. At a minimum having these values on the status page as line items would be a good start.

jgyates, 2020-01-09:
The maintenance page shows the following:

    kW Hours in last 30 days : 
    Fuel Consumption in last 30 days : 
    Total Power Log Fuel Consumption : 
    Run Hours in last 30 days : 

I have opted not to include this info in a graph for performance reasons (i.e. there is a lot of data to transfer for the graph on top of the kw log data currently being used).

jgyates, 2020-01-09:
If yours does not show this let me know.

Appfrontier, 2020-01-09:
> The maintenance page shows the following:
>
> ```
> kW Hours in last 30 days :
> Fuel Consumption in last 30 days :
> Total Power Log Fuel Consumption :
> Run Hours in last 30 days :
> ```
>
> I have opted not [...]

Mine does not show that info:

[image]

jgyates, 2020-01-09:
If you have outgoing mail setup you can send me your logs on the About page. With this data I can likely determine why this is occuring. My guess is that it is due to a logic error. NG systems do not show the fuel gauge since this only applies to stored fuel, but they can still show how much fuel is used.

Appfrontier, 2020-01-09:
> If you have outgoing mail setup you can send me your logs on the About page. With this data I can likely determine why this is occuring. My guess is that it is due to a logic error. NG systems do [...]

Done

jgyates, 2020-01-09:
I checked in an update that should fix this issue. The problem caused the fuel stats listed above not to be displayed for NG systems (or any system with a tank sizes of zero, NG systems do not allow the tank size to be modified). This would also prevent fuel calculation from showing up in the outage log for NG system. Let me know if you have any issues.

Appfrontier, 2020-01-09:
> I checked in an update that should fix this issue. The problem caused the fuel stats listed above not to be displayed for NG systems (or any system with a tank sizes of zero, NG systems do not allow [...]

That fixed it. Thanks

closed 2020-01-09 by jgyates