jgyates/genmon #342: Unauthenticate SMTP Feature Request

zbourdeau, 2020-01-10:
First off, this is a great effort. Thank you for putting this together for the community! I have been using your genmon project successfully with a [https://www.usriot.com/products/serial-wifi-converter.html](USR serial to Wifi convertor ) using the TCP /IP option with great success.

I would like to respectfully request the addition of a boolean value to allow an unauthenticated SMTP server. For those setups where an on premise SMTP server may be present this would be very helpful.

- Generator Model: Generac 17Kw- 2016 model
- Genmon Version: V1.13.33

zbourdeau, 2020-01-10:
@jgyates can you add me as a collaborator on this project? I created a branch for the unauthenticated SMTP change and got it working and would like to submit a PR for you to take a look at.

jgyates, 2020-01-10:
you are now a collaborator.

zbourdeau, 2020-01-10:
It looks like I am still missing permissions to push/ create a branch:

Opening repositories:
S:\Sources\genmon
Pushing smtpauthoption
Error encountered while pushing branch to the remote repository: Git failed with a fatal error.
unable to access 'https://github.com/jgyates/genmon/': The requested URL returned error: 403
Pushing to https://github.com/jgyates/genmon

I'm not sure what is up. Here is a link to a document which may help though: https://help.github.com/en/github/setting-up-and-managing-your-github-user-account/permission-levels-for-a-user-account-repository

jgyates, 2020-01-10:
You have to accept the collaborator invitation.

[image]

jgyates, 2020-01-10:
you should have received a link via email. You may check your spam folder.

zbourdeau, 2020-01-10:
Ack. I accepted and added my PR. Please review it.

jgyates, 2020-01-10:
Looks great! It is merged. The only thing I added was I created a tool tip for the setting, all other changes were taken unchanged. Thanks for your help!

closed 2020-01-10 by jgyates