jgyates/genmon #344: Openweather API Rain total issu

Appfrontier, 2020-01-11:
The OpenWeather API seems to only return rain for the last 1h

The monitoring always shows Rain last 3 hours as 0

[image]

When making a call to the api as follows:

api.openweathermap.org/data/2.5/weather?id=5161911&APPID=XXXXX

The returned JSON for the rain object is as follows

[image]

The docs show that either could be returned. Not sure why it is showing rain for 1h but 0 for 3h as shouldn't the 3h total include the 1h total lol...

The docs show:

[image]

So I guess it would be nice to see both the 1h and the 3h as the docs state if the event has no data it will not be shown hence why 3h is missing.

jgyates, 2020-01-11:
I checked in V1.13.36 that should fix this problem. Good find. Thanks for the feedback. In testing it looks like this data can be returned from the openweather servers as only 1h. The code is now setup to show 1h and 3h if both are returned from the server. In my testing I am only seeing the 1h returned. Let me know if you have any issues or questions.

closed 2020-01-11 by jgyates

Appfrontier, 2020-01-11:
Thanks. May be a small bug but each one is an opportunity to review code :)

JeanB51, 2020-01-11:
FWIW, I see on the openweathermap site, that you can either query 4day-hourly forecasts, or 5-day/3-hour forecasts...maybe this is why only one of the fields is filled in ?
https://openweathermap.org/api

Jean

    Le samedi 11 janvier 2020 13 h 49 min 52 s UTC−5, Appfrontier <notifications@github.com> a écrit :  
 
 
Thanks. May be a small bug but each one is an opportunity to review code :)

.

Appfrontier, 2020-01-11:
The 5 day forecast does indeed return 3h rain (forecast) for each 3 hour time block. I would assume the hourly forecast would return the 1h but cannot prove it since my API key does not allow access to the pro endpoint.