jgyates/genmon #345: Generator monitor display - numbers often unreadable due to formatting

DC072012, 2020-01-13:
Use the template below if you have an issue or want to report a bug. If you have a question or a feature request you can ignore the questions below.

***NOTE:*** If you are having issues with your serial connection, please read this page before posting:

https://github.com/jgyates/genmon/wiki/3.6---Serial-Troubleshooting

If you are having other issues, please see the following page:

https://github.com/jgyates/genmon/wiki/3.5---General-Troubleshooting

If you need to send you logs and registers to the developer, if you email is setup and working properly you can click send your logs on the About page in the web interface.

### Expected Behavior

{Please write here}. Voltage numbers, frequency, etc will display within the “box with rounded corners” for each field.

### Actual Behavior

{Please write here}. Often the numbers display between boxes and are not readable.  Not a huge issue but might be easily fixed in a future release.

### Steps to Reproduce (including precondition)

{Please write here} Happens often but seemingly randomly on iPhone 6 plus and 7 plus phones when phone is held in portrait mode, and switching between menu items (such as status and monitor).  Sometimes turning phone to landscape mode, then back to portrait mode displays the numbers correctly for a while.

### Screenshot or Pictures relating to the problem (if possible)

[image]

{Please write here}. Including screenshot off iPhone 7 plus when numbers are not displayed correctly.

### Your Environment

- Generator Model: {Please write here} 5882
- Generator Registers: Submit via the About page
- Genmon Version: {Please write here} 1.13.32.

jgyates, 2020-01-14:
I don't have an iPhone 6 or 7 to test with so I will have to install a simulator to try to duplicate this. A few questions:

What iOS version?
I assume this is safari? what version?
How much memory does your device have? 
You mentioned iPhone 6 plus and 7 plus in your post, have you duplicated this on both?
does a reboot of the phone change the frequency or your ability to duplicate this?

I don't know if Safari does this but Firefox will reduce some features / functionality on some systems if certain systems resources are not met (i.e. CPU and memory requirements)

DC072012, 2020-01-14:
Iphone 7 Plus is almost exclusively where I’ve seen the problem, mainly because that’s the phone I use the most.  So I’ll focus on that phone.  Model MN5V2LL/A.
IOS 13.3, 128GB storage, 86GB of storage free.  I should have plenty of storage.

I rarely reboot the phone because I generally don’t need to. 
I use Safari exclusively.  Don’t know what version it is, Apple doesn’t report that, only the IOS version.
The main pattern for the artifact to show up is switching from “Monitor” to “Status” which is a pattern I do frequently.  

In fact, I just did that while typing this note.   Switched from Monitor to Status and the numbers are off like in the photo.  Rebooted the phone.  Selected Monitor, then 
Status, numbers are off.  Monitor then status, numbers still off.  Notifications then Status, numbers still off.  About then Status, numbers still off.  
Once the numbers are off the only way to fix it is to turn the phone in landscape mode, then back to portrait mode.  Numbers then look normal within the boxes.
But then selecting Monitor then Status, numbers are off again.

Thanks

liltux, 2020-01-14:
@jgyates I have an IphoneXR and I am having similar results, if not same as @DC072012.  I experience the same offsetting within Safari or Google Chrome browser.  If offset occurs, a refresh/reloading of the page corrects.  Not sure if that helps but more info.  Like the page is not adjusting correctly between some of the pages?

jgyates, 2020-01-14:
I can duplicate this with the iOS simulator. I will let you know once I have a fix.

jgyates, 2020-01-16:
I have checked in V1.13.36. This is a workaround solution. It is better but not quite the solution I would like since I still don't know the exact cause. The workaround should display the gauges text within the gauge border. I am aware that in some instances the text may not be uniformly aligned but until I determine the exact cause of this issue. The code will detect if the environment is mobile safari and apply the workaround. Let me know if there are any issues or questions.

DC072012, 2020-01-17:
So far so good!  Excellent work.  

How would I make a contribution towards Genmon?  I don’t see a way to do that, maybe I’m not looking at the correct page.

jgyates, 2020-01-17:
This link has a donate button for paypal:

https://github.com/jgyates/genmon/wiki#support

Thanks!

closed 2020-01-23 by jgyates