jgyates/genmon #346: UI Color meanings informational - NOT and Issue

wotseurba, 2020-01-16:
Sorry to ask this, but I couldn't find an explanation in documentation.

Why are my menu buttons on left side of UI a Gold color?

I remember them being Green, and sometimes Red when I had early installation serial faults, but recently the buttons have been this Gold color.  All functions work properly.  And reason for the Gold color?

jgyates, 2020-01-16:
No problem:

Gold / Yellow - Service Due
Green - Switch in Auto  / Ready
Red - Alarm 
Blue - Exercise
Black - Switch Off 
Gray - Running in Manual 
Dark Green (Blue / Green ) Running 

I will add this to the wiki

jgyates, 2020-01-16:
https://github.com/jgyates/genmon/wiki/1----Software-Overview#genservpy-web-applicationoptional

closed 2020-01-16 by jgyates

wotseurba, 2020-01-16:
Thanks for the awesome support.  This is great info.

I did notice under the Maintenance menu, my generator status:
Service :
        Service A Due : 0 hrs or 06/04/2020
        Service B Due : 200 hrs or 06/04/2022
        Battery Check Due : 06/04/2020
        Total Run Hours : 199
        Hardware Version : V1.06
        Firmware Version : V1.19

I will need to learn how to clear the service needs.  I thought the generator status LED would be Yellow when it needed service.  My LED is Green.

Thanks again, you can close this request

jgyates, 2020-01-17:
It will probably turn yellow soon. Looks like you have 199 hours and your service A is due. Once you hit 200 your alarm may trigger. To clear the Alarm/Warning message, press OFF on the control panel keypad
followed by the ENTER key.  Reset the maintenance counter you do that via the menus on the LCD on the controller (either the Edit or Dealer > Edit)

To access Dealer Menu on the Evolution controller, from the top menu selection (SYSTEM, DATE/TIME,BATTERY, SUB-MENUS) enter UP UP ESC DOWN UP ESC UP, then go to the dealer menu and press enter. For liquid cooled models a level 2 dealer code can be entered, ESC UP UP DOWN DOWN ESC ESC, then navigate to the dealer menu and press enter. Passcode for Nexus controller is ESC, UP, UP ESC, DOWN, UP, ESC, UP, UP, ENTER.