jgyates/genmon #347: Minor nomenclature issue

DC072012, 2020-01-17:
Use the template below if you have an issue or want to report a bug. If you have a question or a feature request you can ignore the questions below.

***NOTE:*** If you are having issues with your serial connection, please read this page before posting:

https://github.com/jgyates/genmon/wiki/3.6---Serial-Troubleshooting

If you are having other issues, please see the following page:

https://github.com/jgyates/genmon/wiki/3.5---General-Troubleshooting

If you need to send you logs and registers to the developer, if you email is setup and working properly you can click send your logs on the About page in the web interface.

### Expected Behavior

My controller showed an error message as "Inspect Battery" and the light on the side of the generator was yellow.  I would expect Genmon to have yellow warning state and "Inspect Battery".  Apparently this is a timed maintenance reminder message that I was able to clear by hitting enter twice on the controller.  

I read through the wiki and didn't see this flagged as a "known issue" although I might have missed it

I don't remember seeing the battery voltage down to 12.4 very often so I'll have to watch that.  It is very cold here today, not sure if that is the reason for the lower voltage, but regardless it's  not a Genmon issue.

### Actual Behavior

Genmon shows red error state, and "Check Battery vs. Inspect Battery which is a yellow state.

### Steps to Reproduce (including precondition)

{Please write here}

### Screenshot or Pictures relating to the problem (if possible)

{Please write here}

### Your Environment

- Generator Model: 5882-0
- Generator Registers: Done
- Genmon Version: V1.13.37
[image]
[image]

jgyates, 2020-01-17:
Thanks for the feedback. 

I just checked in an update (V1.13.38) that will change the wording on the Nexus models from Check Battery to Inspect Battery.

Regarding the color, the software treats all alarms with the same color coding (red) since from the modbus perspective there is no known way to differentiate between a critical (generator will not start) alarm and a warning  (generator will start) alarm.

The red in the menu indicates an alarm, the menu changes to yellow when service is due.

The web interface menu will change colors based on your generators status:

Gold / Yellow - Service Due
Green - Switch in Auto / Ready
Red - Alarm
Blue - Exercise
Black - Switch Off
Gray - Running in Manual
Dark Green (Blue / Green ) Running

swvalenti194, 2020-01-17:
V1.13.38 - 2020-01-17
Changed text to "Inspect Batter" for Nexus models

:-)

jgyates, 2020-01-17:
oops, that was a typo in the changelog. I fixed it and validated that I did not do the same in genmon. Thanks!

closed 2020-01-17 by jgyates