jgyates/genmon #348: [Feature Request] RPM Graph

USAFPride, 2020-01-19:
Most, if not all generators typically run at 3,600 RPM. It may be more helpful if the Graph had a wider spectrum around 3600. eg. 0-1500 would be 25% of the graph, 1500 - 3000 the next 25%, and 3000 - 4000 the remaining 50%

jgyates, 2020-01-19:
@USAFPride, thanks for the suggestion. 

The RPM gauge gradients depend on the generator specs:

60 Hz Diesel models are 1800 RPM
50 Hz Diesel models are 1500 RPM
60 Hz NG / LP models are 3600 RPM
50 Hz NG / LP models are 3000 RPM

Since the rated RPM should not deviate that much regardless of the load, the RPM gauge is color coded based on tolerances (+/- 50  RPM and +/- 75 RPM). The exception to this is quiet mode exercise where RPM (and voltage output) is intentionally reduced.

So for example for a 3600 RPM model, the green area is in a range of 3550 - 3650, the yellow area is in the range of 3525 - 3675 and all other areas are red. The four quarter gradients are just a division of the nominal by 4 (900,1800,2700,3600 for a 3600 RPM generator)

I agree that there is a lot of red on the gauge and a very small amount of green, however this is the linear representation of the tolerances. At some point I want to redo the gauges to a more pleasing aesthetic while keeping the linear representation (I am thinking a gauge that changes color dynamically instead of static colored regions), but outside of changing the gauge control completely then this implementation is the best representation while keeping a linear representation. Just changing the scale around 3600 would result in a gauge that is skewed or nonlinear.

In short, I want to update the gauges at some point, but in my view, just changing the gradient around the nominal would not accurately represent the data and tolerances.

USAFPride, 2020-01-20:
That's probably a better description than I gave. Less Red, more Green.

closed 2020-01-20 by jgyates