jgyates/genmon #349: Enhancement request - user selectable battery charge level trigger

DC072012, 2020-01-20:
Use the template below if you have an issue or want to report a bug. If you have a question or a feature request you can ignore the questions below.

***NOTE:*** If you are having issues with your serial connection, please read this page before posting:

https://github.com/jgyates/genmon/wiki/3.6---Serial-Troubleshooting

If you are having other issues, please see the following page:

https://github.com/jgyates/genmon/wiki/3.5---General-Troubleshooting

If you need to send you logs and registers to the developer, if you email is setup and working properly you can click send your logs on the About page in the web interface.

### Expected Behavior

It would be very helpful if there was an ability in Genmon to set a threshold battery voltage trigger, and if the battery discharges below that voltage send a notification via the established notification process.

I realize the Nexus controller will throw a "low battery" error but that does not occur until the battery is 10.8 volts, which is a voltage so low that damage to the battery can already be occurring.  Based on the experience of several auto mechanics I know, if a battery's voltage drops below 11.5 volts the plates start to degrade.  It's especially a problem in the area of New York I live in where we have very cold weather.  The colder the battery, the harder it is to start a cold engine with a potentially diminished battery.

I couldn't find anything where this had been talked about before.  Thank you for your consideration.

### Actual Behavior

{Please write here}

### Steps to Reproduce (including precondition)

{Please write here}

### Screenshot or Pictures relating to the problem (if possible)

{Please write here}

### Your Environment

- Generator Model: {Please write here}
- Generator Registers: Submit via the About page
- Genmon Version: {Please write here}

jgyates, 2020-01-20:
According to this doc the alarm is triggered at 11.9V. See page 8-5

http://soa.generac.com/manuals/6219710/CNTRLNEXUS001

DC072012, 2020-01-20:
OK, thanks for the reference.  A Generac tech who worked on my unit gave me the 10.8 volt threshold number for the low battery warning.  Apparently he was mis-informed.  11.9 volts should be OK.

closed 2020-01-20 by jgyates