jgyates/genmon #350: New Model, no open molex port on board

TikiGuy, 2020-01-23:
I set up an 11KW model a few years back with zero issues.  Moved into a new house and had a 16KW model installed and the connection point under the sticker appears to not have the molex connector soldered in.  It looks like port 1 is the proper connection type but is already in use.

[image]

liltux, 2020-01-23:
@TikiGuy does your unit have an Evolution 2 controller?  If so that is for the mobillink wifi module.  You will have to either remove the mobile link and go with yours or use a splice cable that will allow both to be connected.

TikiGuy, 2020-01-23:
Yeah Evolution 2 controller.  Guess they connect to it with their included wifi now.  I'll unplug that one and give it a go later tonight.  Thanks for the quick reply.

TikiGuy, 2020-01-23:
I was able disconnect the right connector in the picture above and plug in there.  Didn't work until I remembered about shared ground, hooked up to battery for power and it started working.

Thanks for your assistance.

closed 2020-01-23 by TikiGuy