jgyates/genmon #351: Outbound email failure

erfahnoe, 2020-01-28:
Use the template below if you have an issue or want to report a bug. If you have a question or a feature request you can ignore the questions below.

### Expected Behavior

My Genmon was set up to send generator status notices to my gmail.com account. This has been working as expected since last May. I normally get a set of exercise notices to my gmail account every Sunday as scheduled.

### Actual Behavior

No exercise notifications have been received for the last two Sundays. Email tests result in the following error message: 
"An error occured:
Error sending email: (535, '5.7.8 Username and Password not accepted. Learn more at\n5.7.8 https://support.google.com/mail/?p=BadCredentials v18sm5117021qkg.67 - gsmtp')
Please try again."

I confirmed that my gmail login and password were correct before several retests with the same error notice. I then upgraded Genmon to V1.13.38. Everything works except that the email test continues to fail. Also I notice that a new switch 'Disable SMTP Auth' has been added to Outbound Email Settings with the upgrade. I have left this switch at the default OFF position. 

I have received several notices from Google that my email account has rejected email from an unauthorized Linux device. The location cited corresponds with my local FiOS network. I suspect that Google has recently changed the security requirements for SMTP access to gmail accounts.

Any help or guidance you can provide to remedy this outbound email issue will be appreciated.

### Steps to Reproduce (including precondition)

{Please write here}

### Screenshot or Pictures relating to the problem (if possible)

{Please write here}

### Your Environment

- Generator Model: Generac Synergy
- Generator Registers: Submit via the About page
- Genmon Version: V1.13.38

jgyates, 2020-01-28:
I am not aware of any changes that would cause this problem. I use gmail with the latest version with no issues. I would check your gmail setting for "Use less secure apps" to make sure that this setting has not changed on your gmail account. See these links for details:

https://support.google.com/accounts/answer/6010255?hl=en

https://myaccount.google.com/lesssecureapps

The "use less secure apps" setting in gmail should be enabled to allow traditional email clients (i.e. non OAuth clients) the ability to send via gmail. I use a dedicated gmail account for sending email with genmon.

erfahnoe, 2020-01-28:
You are correct, Google had upgraded security on my account recently, the default for Use Less Secure Apps was OFF.  All is working again, I only use my gmail mail account for this type of 'privacy not needed' stuff too. It's very handy when traveling.   
Thank you for your responsiveness to this minor issue. I really appreciate all your effort in developing and supporting Genmon, I love it!

closed 2020-01-28 by erfahnoe