jgyates/genmon #353: [Feature Request] Start & Transfer after x dropouts in a period of time

JMVS, 2020-02-03:
Hi @jgyates.

I know this might be an isolated/edge case but it would be nice if you could implement it.
The electrical service in Argentina is really degraded and in summertime we experience low voltage. I have an autotransformer to cope with that but even sometimes it not enough and we experience short outages as the autotransformer cuts the power when is below the minimum threshold. In turn the generator kicks in and shortly after the voltage comes back to a value the autotransformer can stabilize (it waits for 1 minute before restoring the line power) and the generator goes into cooling down mode and then off. Moments later the same process starts all over again. What I do in these cases is manually start & transfer the gen via genmon's web interface whilst monitoring the utility line status.
Can this process be automated somehow? Like if there are "x" amount of short outages for certain period of time, start & transfer the generator and then wait until the utility power is above a defined level for an "y" amount of time, transfer back to utility, cool down, stop and restart the counter. I know...easier said than done...

This also _might_ be doable with something like Home Assistant (that I have running along with genmon's MQTT). Do we have access to Stop/Start/Start & Transfer via MQTT messages? If possible, anyone care to share a possible setup?

Thanks in advance.

jgyates, 2020-02-04:
@JMVS 

Let me think on this. One issue that would need to be thought about is the implementation of the timing: determine how long a low line voltage would need to be low before doing anything. Off the top of my head the parameters could be:

1) Threshold Voltage - If Utility drops below this value then we pay attention 
2) Pickup Voltage - If Utility if above this it is normal 
3) Minimum Brownout Time - Utility must be below threshold for this period of time to be considered an outage 
4) Minimum Restore Time - Utility must be above the pickup voltage for this period of time to be considered an end of an outage 
5) Number of Outages- Number of outages before "Start and Transfer" is issued 
6) Amount of time without an outage that has to pass before stopping a previous "Start and Transfer"

Does this sound like what you are talking about?

JMVS, 2020-02-04:
@jgyates Yup. That pretty much covers it.
In my case if in the span of an hour I get 3-4 short outages* I then trasfer to gen's power for an hour (I'm obviously not constantly monitoring the utiliy voltage) and then back to utility power. Fuel consupmtion should be accounted for.

*short outages: Please note that this is due to my setup. If I remove the autotransformer from the equation, what I would experiencie would be a _dangerous_ dropout in voltage and not an actual outage.

skipfire, 2020-05-24:
I have similar issues in Lousiana where we will repeatedly lose power for shorter durations.  If the full feature described above is problematic, what about the existing "Start & Transfer" followed by options to "allow return to utility in (x) minutes".  Then I could have it go to the generator and know it will say there for the next (x) minutes (say 180), and then once the 180 minutes is up it will follow normal rules for returning to the utility.

jgyates, 2020-09-29:
I am tracking the feature request list here: https://github.com/jgyates/genmon/blob/master/feature_requests.md

I will be closing this thread for now but the request is still on the master request list.

closed 2020-09-29 by jgyates