jgyates/genmon #354: Verify Setup

farberm, 2020-02-15:
1. I am not receiving any data from the generator to the Pi. I have tested the cable and it passes without any issues. I suspect it may have to do with the grounding setup. Could you please verify that I have this correct. 
2. If my ground wiring is correct,  do I have to have wifi turned on to receive data. I currently have not set the wifi on the generator up at all?

Wiring Setup as follows:
GPIO Board to RS232
Pin 4 --> Black Wire --> VCC
Pin 6 --> White Wire --> GND
Pin 8 --> Grey Wire --> TX
Pin 10 --> Purple Wire --> RX

This the the board I ordered: https://www.amazon.com/gp/product/B01JYNHNW6/ref=ppx_yo_dt_b_search_asin_title?ie=UTF8&psc=1

Generator Battery --> Motorcycle USB Power cable (Red positive.Black Ground)--> Power to PI

Generator Battery GND --> Clear Wire --Breakout DB-9 Pin 5

Molex 8 --> Red Wire --> Breakout DB-9 Pin 2
Molex 7 --> Black Wire --> Breakout DB-9 Pin 3
[image]
[image]
[image]
[image]

### Actual Behavior
No data passed from Generator
### Steps to Reproduce (including precondition)
Tested the serial connection and it passes. I have tested the cable all the way to the molex connector.

### Screenshot or Pictures relating to the problem (if possible)

See attached.
Generator Evolution 2

- Generator Model: Evo 2
- Generator Registers: 
- Genmon Version: 1.13.38

jgyates, 2020-02-15:
I can't make out the pin connections on your molex to verify the pin numbers are correct from the picture.

Try swapping pin 2 and 3 on the green breakout connector. If the loopback works and the ground is good this would be the next thing to try.

The pinout for a male vs female DB9 connector is different (pins 2 and 3 are reversed depending on if the connector is male or female). Assuming you have a female DB9 you should be using the lower example in this picture.

https://raw.githubusercontent.com/jgyates/genmon/master/Diagrams/Cable_Connections.jpg

Also, the clear wire does not appear to be going to the ground pin on the green breakout connector (unless there are two ground pins).

farberm, 2020-02-15:
Here are some more pictures.

Red Wire is going to Pin 2, Red wire going to Molex Pin 8 top left on picture
Black Wire is going to Pin 3, Black wire going to Molex Pin 7 second to the left on the top row

Here is a picture of my RS 232 Female correct?

**********
Clear wire may be the issue. I have it going to Pin 5 on the DB-9... I realize that the GND is listed on the other side of the DB-9 but Pin 5 is GND on the RS 232 according to your diagram? Am I misreading which GND you are referring to?

 Is this my problem?
********

jgyates, 2020-02-15:
Yes, pin 5 is ground on the RS-232 DB-9 connector. If you screw down connector labeled 5 goes to ground you should be OK.

The pictures get stripped off if you send them as attachments. You have to attach them from the github web interface.

jgyates, 2020-02-16:
Were you able to get this working?

farberm, 2020-02-16:
Will reinstall tomorrow

jgyates, 2020-02-24:
I am closing this issue. If you have any other questions on this topic feel free to post them to this thread.

closed 2020-02-24 by jgyates