jgyates/genmon #355: Outage log not showing on web page after restore from backup

hokie21, 2020-02-16:
I upgraded the OS on my Pi Zero W. Before doing this I used the "backup" button to back up my Genmon configuration. Everything worked fine, except now my outage log shows nothing. my outage.txt file in in /etc/genmon and has six entries which are not showing up on the Outage/Outage Log page.  How do I make these entries show up on the web GUI?

jgyates, 2020-02-16:
To restore you should do it in this order:

1) You should stop genmon:  ./startgenmon.sh stop
2) copy the files to /etc/genmon 
3) restart genmon: ./startgenmon.sh restart

If you attempt to restore the files while genmon is running you could see issues.

Unless you change the user that genmon runs (root) you should not have a permission issue.

hokie21, 2020-02-16:
I believe I copied the files over before I started it, but I don't remember if it started automatically after the installation completed. 

Now I've restarted genmon from the command line. I didn't change the genmon user from root. I'm still not seeing the outage data in the UI. I changed some config parameters through the UI and see that the files updated, so it doesn't look like a permission issue. I see no reason why the program cant read the outage.txt file

jgyates, 2020-02-16:
have you validated the contents of the file /etc/genmon/outage.txt ?

       cat /etc/genmon/outage.txt 

You should see the outage info listed, one line per outage.

Then let's look at the permissions:

      ls -la /etc/genmon/outage.txt

hokie21, 2020-02-16:
pi@raspberrypi:/etc/genmon $ cat outage.txt
2019-08-07 14:34:40,1:08:07
2019-08-08 16:58:03,0:00:00
2018-11-26 00:04:31,0:00:34
2018-11-26 02:27:06,0:00:00
2018-11-26 06:53:31,0:00:34
2018-11-26 06:54:07,0:01:36

pi@raspberrypi:/etc/genmon $ ls -la outage.txt
-rw-r--r-- 1 pi pi 168 Feb 14 20:08 outage.txt

jgyates, 2020-02-16:
I have tried to duplicate but the copy works fine for me.  Are you using the latest version? You could attach your outage.txt file and I could try to import yours into my test setup.

hokie21, 2020-02-16:
I am using V1.14.02.

[outage.txt](https://github.com/jgyates/genmon/files/4211049/outage.txt)

jgyates, 2020-02-16:
It works for me with your file:

[image]

I copied it using this command:

     sudo cp outage.txt /etc/genmon/outage.txt

hokie21, 2020-02-17:
very strange. I tried dos2unix, copied it back over using scp, restarted and am still seeing this

[image]

jgyates, 2020-02-17:
If you run dos2unix on the file it may make it unreadable for the program. I have never tried that. you should not have to use dos2unix on the file since it was created on linux and is being read on linux.

hokie21, 2020-02-17:
I went back to the original outage.txt file that had not been dos2unixd and am still seeing the display above.

jgyates, 2020-02-17:
That is odd. If you have outgoing mail enabled you can send me your log files and I can see if anything odd is in your error log. You can send the logs from the About page in the web interface. You can also check out /var/log/genmon.log

hokie21, 2020-02-17:
Logs sent. I don't see anything unusual in genmon.log. Thank you.

jgyates, 2020-02-17:
I got your logs. I don't see any errors. have you tried rebooting your pi?

     sudo reboot now

Are you using command line tools to copy the file over? for example when I copy the file you sent I copied it to my mac, then used this command to copy the file from my Mac to the /home/pi directory on my pi:

     scp outage.txt pi@192.168.1.10:./

Then I logged into the pi (/home/pi directory) and copied the file to the /etc/genmon directory using sudo .

    sudo cp ourtage.txt /etc/genmon

hokie21, 2020-02-17:
The path to the outage file in genmon.conf was incorrect. I fixed that and restarted from the command line. It's working now. Thanks for your help!

jgyates, 2020-02-17:
No problem, I am glad it is working for you.

closed 2020-02-17 by jgyates