jgyates/genmon #356: MQTT integration

farberm, 2020-02-16:
### Behavior
I have turned on JSON numeric option. This allows some of the MQTT information to have values in the topic (Utility Voltage, Battery Voltage). However there is not calue topic for Ambient Temperature for instance, just generator/Maintenance/Ambient Temperature Sensor, not generator/Maintenance/Ambient Temperature Sensor/value. This result in an error when trying to read subscription...error . Temp: update unable to convert '42 F' to float: invalid literal for float(): 42 F

i do nothave this issue is the topic have a value subscription like generator/Status/Line/Utility Voltage/value

- Generator Model: Evolution 2
- Generator Registers: Submit via the About page
- Genmon Version: 1.14

Is there anyway to send values for all topics not just some of them?

jgyates, 2020-02-16:
The JSON numeric option will return numeric values for data on the status page, not the other pages. the ambient temp is on the Maintenance page. The data on the Maintenance page will return string data. This is the expected behavior. This is the tool tip data for the "JSON for Numerics" option: "If enabled will return numeric values in the Status topic as a JSON string which can be converted to an object with integer or float values."

farberm, 2020-02-16:
It there anyway to get values for the other pages?

 The maintenance page is returning string information as you have said but not payload information. Is there a way to send payload information instead of strings?

jgyates, 2020-02-16:
I can add this to the list.

farberm, 2020-02-16:
That would be great

liltux, 2020-02-16:
@farberm what I have done to get data in the right format you can setup an mqtt server and nodered on the same pi that is running GenMon.  Then send GenMon mqtt data to localhost, using nodered to subscribe to the data, make any changes to data such as convert string to number, and send data either to another mqtt server or other services.   I would recommend a Pi3 or better.

farberm, 2020-02-16:
Thanks for the alternative to waiting for an update. I have a few other things to tackel right now so it will be a little bit before I can get to setting that up the workaround.

jgyates, 2020-02-18:
I have check in V1.14.03. This adds the "JSON for Numerics" option function on the Status, Maintenance, and Outage pages. Note this applies to Evolution and Nexus controllers. Industrial controllers still only respond to the "JSON for Numerics" on the Status page. Let me know if there are any questions.

farberm, 2020-02-18:
Awesome

jgyates, 2020-02-18:
I am going to close this thread for now but feel free to post here if you have issues or questions.

closed 2020-02-18 by jgyates