jgyates/genmon #357: Incorrect Generator Model detected

armhoge, 2020-02-21:
## Incorrect Generator Model detected
### Expected Behavior
Correct model detection and KVA settings
model = G0072440
nominalkw = 7.5 KVA

### Actual Behavior
Generator model reported
model = G0062770
nominalkw = 20

### Steps to Reproduce (including precondition)

Nothing special. Install went perfectly. Only items relating to incorrect model seem to be affected.
GREAT product

### Screenshot or Pictures relating to the problem (if possible)
pramac.jpg

### Your Environment
- sitename = Springbok 
- Generator Model: G0072440 (Pramac sold as Generac)
- Generator Registers: has been sent via the About page. No feedback as to where it went.
- Genmon Version: V1.14.03
[image]

jgyates, 2020-02-21:
I will need your logs and registers to look into this issue. You can submit them if you have outgoing email enabled from the About page on the web interface.

armhoge, 2020-02-22:
Collected regs using ClientInterface.py (genpi_allregs_20200222_0758.gz)
Collected logs defined in LogList FilesToSend (genpi_logs_20200222_0758.gz)

[genpi_allregs_20200222_0758.gz](https://github.com/jgyates/genmon/files/4238898/genpi_allregs_20200222_0758.gz)
[genpi_logs_20200222_0758.gz](https://github.com/jgyates/genmon/files/4238899/genpi_logs_20200222_0758.gz)

jgyates, 2020-02-22:
When I test in my simulator with your registers it works for me. 

The problem could be that your initial install had a serial data error or could not access the internet during the initial run of the program. If the serial device was not acting properly then this could cause an invalid value for register 0019, which could cause the software to misidentify the generator. This is just a guess.

The model lookup takes place on the first run of the program. You can force an re-run of the model detection by deleting the lines that start with 

version = 
and 
model = 

in /etc/genmon/genmon.conf, then restart the software:

/home/pi/genmon/startgenmon.sh restart

When I run the program with your registers my log shows this:

2020-02-22 15:02:01,335 : Looking up model info on internet
2020-02-22 15:02:11,165 : Found: Model: G0072440
2020-02-22 15:02:11,166 : Notice: product ID not found online. Using serial number to complete lookup. : generac_evolution.py:757

Results are:
Model : G0072440
Controller Detected : Evolution 2.0, Air Cooled

armhoge, 2020-02-24:
Did as requested first two times I had a timeout

2020-02-24 16:29:22,364 : Generator Monitor Shutdown
2020-02-24 16:29:24,442 : New version detected: Old = V1.14.03, New = V1.14.03
2020-02-24 16:29:24,452 : GenMon Loaded for site: Springbok
2020-02-24 16:29:27,475 : Looking up model info on internet
2020-02-24 16:29:38,748 : Error in LookUpSNInfo (request 1): ('The read operation timed out',) : generac_evolution.py:731

Then on subsequent cycle gen was detected as shown in your tests

2020-02-24 16:32:49,905 : Generator Monitor Shutdown
2020-02-24 16:32:51,986 : New version detected: Old = V1.14.03, New = V1.14.03
2020-02-24 16:32:51,996 : GenMon Loaded for site: Springbok
2020-02-24 16:32:55,018 : Looking up model info on internet
2020-02-24 16:32:56,406 : Found: Model: G0072440
2020-02-24 16:32:56,406 : Notice: product ID not found online. Using serial number to complete lookup. : generac_evolution.py:757

armhoge, 2020-02-24:
Thank you for your prompt help. 
Closing issue as resolved

closed 2020-02-24 by armhoge