jgyates/genmon #358: Scroll bars with magnification

GlennGoddard, 2020-02-28:
### Expected Behavior

Pages that have more info then will fit on screen should have scroll bar that scrolls.

### Actual Behavior

Scroll bar is non-functional, only items that fit on current page are visible and accessible.  Specifically on the Add-On page

### Steps to Reproduce (including precondition)

My main monitor is at 150% and when opening up GenMon in FireFox on my main monitor I cannot scroll down.  If I drag the page to a non-magnified monitor, everything displays, and then drag it back scroll bars are present.  GenMon is the only time I have experienced this, this started a few months ago after one of the updates; I could not tell you which one.

I would have submitted earlier but until recently I didn't realize that it was the magnification that was the problem.

### Screenshot or Pictures relating to the problem (if possible)

{Please write here}

### Your Environment

- Generator Model: Generic Air Cooled 22KW
- Generator Registers: Submit via the About page
- Genmon Version: 1.14.03

jgyates, 2020-02-28:
I am a bit confused regarding the problem. You mention that your "main monitor is at %150". The way you word this it sound like this is a function of your monitor and not your browser. You also mention that when you drag the page back (I assume you are dragging the browser application to a second monitor.  This leads me to assume that this is a dual monitor setup and you have one monitor set via the operating system to magnify the display, correct (i.e. not just zooming to 150% within the browser)? If this is the case, then what is the operating system and version of the operating system. 

Assume my above assumptions are correct about your issue, can you duplicate this with a different browser or is the specific to Firefox.

jgyates, 2020-02-28:
In addion to your operating system I will need to know if this is the 32 bit or 64 bit version of the OS and Firefox and how much memory you have to attempt to duplicate this. Firefox will revert some of the browsers functionality (i.e. reduce features) if you are running less than the ideal amount of memory.

GlennGoddard, 2020-02-28:
I have a multi monitor system; picture attached to explain magnification; it is actually font size.
[image]
[image]

It is windows 10 64 bit, and it is a windows 10 problem, not your program (at least I don't think so).  It is just interesting that it used to work just fine without my tweak.  I cannot reproduce this with any other site.  The problem is reproducible on the GenMon page on other browsers.
Firefox 73.0.1 64bit version
Exact same behavior with Edge browser.
Edge 44.18362.449.0
8 gig or Ram installed, 61% utilized at this last attempt.

I do have a 'work around' but others might have this issue and it is good info for someone.
I did not expect your response to be so quick, I was about to post about MQTT, which I should have started with.

[image]
[image]

GlennGoddard, 2020-02-28:
The two last screen shots above are reversed.  The bottom is the initial view I get which has no ability to scroll down.  The one above that is after I move it to a monitor without magnification and then back.

jgyates, 2020-03-01:
Thanks for the info.

closed 2020-03-01 by jgyates