jgyates/genmon #359: Cannot communicate with GenPi wifi to configure so that I may communicate with my new liquid cooled 24kW Generac generator

jnyikos1, 2020-02-29:
Use the template below if you have an issue or want to report a bug. If you have a question or a feature request you can ignore the questions below.

***NOTE:*** If you are having issues with your serial connection, please read this page before posting:

https://github.com/jgyates/genmon/wiki/3.6---Serial-Troubleshooting

If you are having other issues, please see the following page:

https://github.com/jgyates/genmon/wiki/3.5---General-Troubleshooting

If you need to send you logs and registers to the developer, if you email is setup and working properly you can click send your logs on the About page in the web interface.

### Expected Behavior

{Please write here}

### Actual Behavior

{Please write here}

### Steps to Reproduce (including precondition)

{Please write here}

### Screenshot or Pictures relating to the problem (if possible)

{Please write here}

### Your Environment

- Generator Model: {Please write here}
- Generator Registers: Submit via the About page
- Genmon Version: {Please write here}

liltux, 2020-02-29:
Email liltux at liltux dot com.

jnyikos1, 2020-02-29:
Expected Behavior
I should have been able to easily communicate with GenPi wifi unit installed correctly on generator so that I may configure it to connect to home wifi AP.

Actual Behavior
As soon as I got to step 5 of the instructions, which is go to wifisetup.pi:1880/GenPi and type in my home SSID and password, the GenPi disappeared from any network, and never showed back up. It stated "connection lost" on this settings screen and I was not able to proceed in the setup / config process.

Steps to Reproduce (including precondition)
I went to local wifi GenPi unit I installed on generator and pressed reset button several times and attempted to "get into" this device via my Iphone, and even went through the router in the home and never saw it so that I could get in and configure. I just can't get to the point where I can configure...and even when I do, I feel I will need an available static IP address to complete this process.

Screenshot or Pictures relating to the problem (if possible)
{Please write here}

Your Environment
Generator Model: Generac 24kw liquid cooled
Generator Registers: Submit via the About page
Genmon Version: {Please write here}

liltux, 2020-02-29:
Can you see the genpi access point with a computer or smart phone?  What are the SSID and Password you are trying to use?  Avoid spaces and \, ' or " quotes as they will not interpret correctly in the script.  Also, do you have a good wifi signal at the generator?  Preferred full bars.  Also once it connects to your net work you will have to use your router or fing app to find the IP of the device on your Network.

jnyikos1, 2020-02-29:
We actually saw the IP address with our Iphones...but did download the Fing App and never saw the Gen Pi...we have 2 out of 3 bars usually, but I can take the operation local to the Genny so I have full bars just to configure the device to communicate with the home AP. There are no spaces or special characters in our SSID or password

jnyikos1, 2020-02-29:
Everytime I pressed the reset button I waited the full 30 seconds...it always showed back up in our que. We can see the generator wifi signal and connect to it no problem. We can get to the wifisetup.pi:1880/GenPi no problem...it is only after we enter the SSID and password to the home wifi that the GenPi and hit submit that it disconnects itself from what we can see on our wifi. Very frustrating!

liltux, 2020-02-29:
Great, so when you put in the SSID AND password, verify spelling.  When you click the save/submit button it runs a script to configure the settings and then reboots.  After reboot it will connect to the network if in range.  At that point you will have to use fing or your router to find the device on your Network. it may take a few seconds to find it.  Be sure your device is connected to the correct network you are searching the device for.

jnyikos1, 2020-02-29:
Sorry, there is a dash in the SSID... "-"

liltux, 2020-02-29:
@jnyikos1 a dash should be fine

liltux, 2020-02-29:
It will say disconnected as it is now connecting to your house wifi and will have a different up address that you will have to identify.  Was the blue led lit?

jnyikos1, 2020-02-29:
Could I use a laptop, a new up to date one with wifi, once it reboots and connects to the home AP so that I can configure it? That is where my hold up is

jnyikos1, 2020-02-29:
ok, so is it a revolving IP address?

jnyikos1, 2020-02-29:
Shouldn't the home router AP be able to identify this unit? This particular end user has about 23 items using different IP addresses on his router.

jnyikos1, 2020-02-29:
blue led is lit...no problem there. Seems like after I enter the SSID and password and it reboots, I need to use either the router to find it, or smart phone so that I may get the most recent IP address it is using? what about subnet mask?

liltux, 2020-02-29:
Yes his router should see the deviyifnit is connected

liltux, 2020-02-29:
If Led is lit then it is connected to the router, no need to hit reset

jnyikos1, 2020-02-29:
My problem is that sometimes when one of these 23 devices connected to his router isn't labled with an IP address I don't have anything that says "genpi"

jnyikos1, 2020-02-29:
I apologize for so much traffic, but you are the most help I've had on this issue for a long time now. I just want the darn thing to act like my Furbo at home, and use the home AP so that I can be in another state and press an App button on personal phone and see what this generator is doing.

liltux, 2020-02-29:
It uses DHCP so all you need is IP. 
http://(new IP address):8000 for GenMon
http://(new IP address):1880/GenPi for GenPi settings.

liltux, 2020-02-29:
You can try the free fing app on your phone

jnyikos1, 2020-02-29:
We had Fing, and we never really saw anything that said "genpi"...is there another identifier? just checking and gathering as much info as I can before I go back to the job site next weekend and try this again.

liltux, 2020-02-29:
It is not a revolving ip, it uses the one issued by the router

jnyikos1, 2020-02-29:
Ok, I will take all this info and see if I can make this all communicate correctly next week. So, I will go to wifisetup.pi:1880/GenPi then I will enter my SSID and password and when it reboots...it is communicating with the home router AP. I will need to utilize Fing to find the IP address that home router gave the GenPi and I will somehow get into the GenPi and configure at that point?

liltux, 2020-02-29:
Yes, but the GenMon runs on :8000 of the IP address to configure.

jnyikos1, 2020-02-29:
My understanding is the 1800 is just the connection point...and then the 8000 is the Genmon software? Am I over thinking?

liltux, 2020-02-29:
Yes

jnyikos1, 2020-02-29:
thank you very much for all your help...I can't believe this kind of support is happening on a Saturday and this fast! I may reach out next weekend if I run into a snag, but I believe my self to be better prepared now for the task.

closed 2020-03-08 by jgyates