jgyates/genmon #360: Maintenance Nominal RPM and Rated kW showing wrong values

mxplea10, 2020-03-01:
Use the template below if you have an issue or want to report a bug. If you have a question or a feature request you can ignore the questions below.

***NOTE:*** If you are having issues with your serial connection, please read this page before posting:

https://github.com/jgyates/genmon/wiki/3.6---Serial-Troubleshooting

If you are having other issues, please see the following page:

https://github.com/jgyates/genmon/wiki/3.5---General-Troubleshooting

If you need to send you logs and registers to the developer, if you email is setup and working properly you can click send your logs on the About page in the web interface.

### Expected Behavior

The Nominal RPM value should be 3600 and Rated kW value should be 22kW

### Actual Behavior

Both the Nominal RPM and Rated kW shows my name as the values.
[image]

Reference [Known Issues #11](https://github.com/jgyates/genmon/wiki/Appendix-D-Known-Issues) "The Evolution 2.0 controller has show that at times it will report what appears to be random data from modbus registers 0001. Genmon will send an email notice if this register changes. Due to what is believed to be a bug in the Evolution firmware there is an option to disable unknown values from being reported in register 0001 for Evolution 2.0 controllers ("Ignore Unknown Values") on the advanced page.": Emails are sent each time there's a setting change.  Is this normal? I can't find the  Advance Page to set Unknown Values.  Where is it located?

### Steps to Reproduce (including precondition)

{Please write here}

### Screenshot or Pictures relating to the problem (if possible)

{Please write here}

### Your Environment

- Generator Model: G0070432
- Generator Registers: Submit via the About page
- Genmon Version: V1.14.03

jgyates, 2020-03-01:
To get to the advanced page, click the gear icon in the top right corner of the web interface twice.

I received your logs. They also indicate that your file /etc/genmon/genmon.conf file has some invalid entries like Nominal RPM and Rated Frequency. I would recommend referting to the default copy of genmon.conf. To do that follow these steps:

1) ssh into the system or log in locally to a console.
2) Stop genmon:

       cd genmon
       ./startgenmon.sh stop 

2) backup your existing genmon.conf 
 
       sudo cp /etc/genmon/genmon.conf /etc/genmon/genmon.conf.bak

3) copy the default genmon.conf to the /etc/genmon folder:

       sudo cp /home/pi/genmon/conf/genmon.conf /etc/genmon

4) Restart genmon. If you had to unplug the pi to log in locally then at this point I would just shut down and connect the pi to the generator and power it up. If you set it up to start genmon on boot it should work. If you logged in via ssh then just type:

       /home/pi/genmon/startgenmon.sh restart 

This will allow the program to start with the generator connected. This will ensure that the model detection is runs as expected.

That should correct your issue. I am not sure why or how your gemon.conf file has the wrong data.

mxplea10, 2020-03-01:
Thanks for the quick response.  With the exception of the Fuel Type, reverting to the default genmon.conf file seems to put in the right values.  I've enabled **Ignore Unknown Values** in the Advance section but I still got 4 to 6 texts while changing the settings even before saving them.

After testing Genmon last week with the Raspberry Pi over WiFi inside the generator cabinet power with the battery adapter that you recommended, I decided to install the Raspberry Pi in my network cabinet indoors.  Thanks for this great program.

jgyates, 2020-03-01:
When the settings are saved the program will restart. You should get two emails on each restart. One just says that the program is starting (category info) and the second one is the initial state of the generator (category infi). Once it has started you will get an email every time the state of the generator changes. You can change the email categories you receive on the Notification page.

I am going to close this issue. Feel free to post related questions to this thread.

closed 2020-03-01 by jgyates