jgyates/genmon #361: Certificate Error for Untrusted Authority Generated When Trying to Access Generator Monitor Securely

mxplea10, 2020-03-02:
Use the template below if you have an issue or want to report a bug. If you have a question or a feature request you can ignore the questions below.

***NOTE:*** If you are having issues with your serial connection, please read this page before posting:

https://github.com/jgyates/genmon/wiki/3.6---Serial-Troubleshooting

If you are having other issues, please see the following page:

https://github.com/jgyates/genmon/wiki/3.5---General-Troubleshooting

If you need to send you logs and registers to the developer, if you email is setup and working properly you can click send your logs on the About page in the web interface.

### Expected Behavior

Use Self-signed Certificate to securely signed into the Generator Monitor.

### Actual Behavior

Since enabling the Webserver Security Settings, the Generator Monitor secure web can't be accessed unless I proceed to the Generator Monitor's IP address which is label as unsafe. The website security certificate isn't trusted whether using Chrome, Internet Explorer or Safari.  The Chrome error is "NET::ERR_CERT_AUTHORITY_INVALID" and Internet Explore error "Error Code: DLG_FLAGS_INVALID_CA" and "DLG_FLAGS_SEC_CERT_CN_INVALID".  I don't have a problem accessing other secure sites.  Is there a configuration change to fix this error or is there some else that I need to do?

Also, is there a way to sign out of the Generator Monitor without having to restart genmon or the device to get back to the login prompt?  Although the browser tab is closed, there is no request for a login when going back to the monitor page.

### Steps to Reproduce (including precondition)

{Please write here}

### Screenshot or Pictures relating to the problem (if possible)

Screenshots for Chrome, Internet Explorer and Safari.
[image]
[image]
[image]

### Your Environment

- Generator Model: G0070432
- Generator Registers: Submit via the About page
- Genmon Version: V1.14.03

jgyates, 2020-03-02:
Unless a give certificate is registered to a given IP address or fully qualified domain name these browsers will give you this warning unless you specifically import the self signed certificate.

https://peacocksoftware.com/blog/make-chrome-auto-accept-your-self-signed-certificate

mxplea10, 2020-03-02:
I'll see if I can figure this out or do something similar with the router.  I tested remote access using DDNS before enabling the Self Signed Certificate;; however, I haven't been able to access it remotely the secure login.  Also, you didn't mention how to properly sign out of the Generator Monitor.

jgyates, 2020-03-02:
To logout you enter this address:

https://YourIPAddressGoesHere/logout

Additional details on the web application are here:

https://github.com/jgyates/genmon/wiki/1----Software-Overview#genservpy-web-applicationoptional

closed 2020-03-08 by jgyates