jgyates/genmon #362: The GenPi is not connecting with local AP router

jnyikos1, 2020-03-07:
### Expected Behavior

Per instructions, after I enter SSID and password on wifisetup.pi:1880/GenPi, the unit disconnects as its' supposed to. Then it should, after a short period of time, reconnect with the AP that I gave the SSID and password to.

### Actual Behavior

GenPi disconnects and then can't be found again on ANY device. I have an Iphone up with FING App and a laptop logged into the actual AP and cannot see the genpi after submitting the SSID and password to the local AP router. 

### Steps to Reproduce (including precondition)

I have been pressing the reset button locally on the GenPi unit several times throughout the course of several hours and going through the instructions step by step very carefully and cannot get this GenPi to connect to the local AP so that we can get the result we are seeking. 

### Screenshot or Pictures relating to the problem (if possible)

Nothing to see.

### Your Environment

- Generator Model: RG02515ANAX
- Generator Registers: Submit via the About page (not sure what this is...it is registered with generac)
- Genmon Version: {Please write here)...I am not sure how I'm supposed to give you a Genmon version when I can't connect to this unit to program it to utilize the local AP router.

jnyikos1, 2020-03-07:
The blue led is lit...and it performs two long and 12 rapid, short blips...does that mean anything? This is so frustrating to be able to communicate with the GenPi after I hit reset button...but then after I give it my SSID and password it disconnects and I can't see it anywhere to identify an IP address.

jgyates, 2020-03-08:
@jnyikos1 

I am going to let @liltux answer this one as this automated wifi code is part of the packaged units that he  supports. I don't have any of that code so I can't really give you any guidance other than to try validate your SSID and password on another device to validate that it is working as expected.

liltux, 2020-03-08:
@jgyates I believe @jnyikos1 contacted me via email and I believe he is up and running.  If he doesn't respond in a day or so you should be ok to close his 2 issues.

jnyikos1, 2020-03-08:
I was able to get it all up and running like normal.
Thank you

closed 2020-03-08 by jgyates