jgyates/genmon #363: Receiving false notifications since Evolution 2.0 controller firmware update

mxplea10, 2020-03-13:
Use the template below if you have an issue or want to report a bug. If you have a question or a feature request you can ignore the questions below.

***NOTE:*** If you are having issues with your serial connection, please read this page before posting:

https://github.com/jgyates/genmon/wiki/3.6---Serial-Troubleshooting

If you are having other issues, please see the following page:

https://github.com/jgyates/genmon/wiki/3.5---General-Troubleshooting

If you need to send you logs and registers to the developer, if you email is setup and working properly you can click send your logs on the About page in the web interface.

### Expected Behavior

{Please write here}

### Actual Behavior

Since the Evolution 2.0 controller received a firmware update, notifications of power outages and alarms are being generated.  There have been no indications of a power outage.  No flickering lights or Mobile Link notifications.  Although "The system is not in alarm", the Engine is "Switch State : System in Alarm" intermittently in the notifications.

Is there a configuration or setting change required for Evolution 2.0 firmware V1.09?

### Steps to Reproduce (including precondition)

{Please write here}

### Screenshot or Pictures relating to the problem (if possible)

[Genmon Notifications.pdf](https://github.com/jgyates/genmon/files/4331869/Genmon.Notifications.pdf)

### Your Environment

- Generator Model: G0070432
- Generator Registers: Submit via the About page
- Genmon Version: V1.14.03

jgyates, 2020-03-13:
The email you sent shows this:

       ...firmware has changed from V1.09 to V0.32.

This is likely not due to a firmware update but rather a data error (i.e. a dropped packet) since the firmware version is lower.  Are you seeing any serial communications errors on the Monitor page under Communication Stats (I am guessing you have timeout errors)? Your log files should show communications errors, if you send me the log files (the other button on the About page) I can validate this. If this was a one time dropout of serial data, the the problem should correct itself. If you continue to see these false state changed then i would look to see if your cable may have loose connections that could be affected by vibrations when the generator is running.

With the generac controllers, if the controller does not respond after a timeout period the next packet will be requested. If the controller responds with the data from the previous packet then the data can get out of sync and some registers can report the wrong value. An outage notice  occurs when the register that reports the line voltage drops below the threshold voltage. If the vale for the line voltage is invalid it would trigger this condition.  Another thing you could try is to increase the modbus timeout on the advanced page. See the setting "Additional Modbus Timeout (sec)". Change this from a zero to 2. That could correct the problem if you are still seeing it.

To validate the theory that this is a data error (first check the Monitor page for communication errors) you can watch the registers change on the registers page (click the gear icon in the upper right once, click it twice for the advanced page). You could see the values change in registers like the version register or the utility voltage register if you are having comm timeouts.

The only Evolution 2.0 specific setting is the "Ignore Unknown Values" on the advanced page. See item 11 here for details on that:

https://github.com/jgyates/genmon/wiki/Appendix-D-Known-Issues

jgyates, 2020-03-14:
You do have serial errors in your log:

These are the ones from the 12th, there are others from earlier:

mymodbus.log:
2020-03-12 18:24:38,218 : Flushing, unexpected data. Likely timeout.
2020-03-12 18:24:38,225 : Error in SendPacketAsMaster: Attempting to use a port that is not open : mymodbus.py:489
2020-03-12 18:24:38,226 : Packet: [157, 3, 0, 83, 0, 1, 105, 135]
2020-03-12 20:00:57,574 : Error: timeout receiving slave packet for register 0238 Buffer:0

myserial.log:
2020-03-12 18:24:38,223 : Error in SerialDevice:Flush : /dev/serial0:Attempting to use a port that is not open : myserial.py:165

I also see serval errors like this one:
Resetting SerialDevice:ReadThread Error: /dev/serial0:device reports readiness to read but returned no data (device disconnected or multiple access on port?) : myserial.py:120

How long has this been occuring? When you set this up did you use serialconfig.py to validate you serial port?

https://github.com/jgyates/genmon/wiki/1----Software-Overview#otherappsserialconfigpy-optional

mxplea10, 2020-03-14:
I have the Raspberry Pi connected via ethernet cable in an indoor network cabinet and the serial cable connected to a molex cable the way that you recommended.  I'll still verify that the cable isn't loose.  This just started Monday or Tuesday and it has been working great for three to four weeks.  If I recall correctly, I didn't use serialconfig.py to configure the serial port because it's stated that it was needed to setup the serial port on the PI 3.  I setup the serial port manually and validated it according to the instructions since I'm using a Raspberry Pi 3 B+.

I also have an UPS monitoring application on this Raspberry Pi 3 B+ and have two UPS plugged into the USB ports. The CPU utilization is low with both applications running.  Is it recommended that Genmon be the only application running on a Raspberry Pi?

mxplea10, 2020-03-14:
Here's the serial port status.
[image]

mxplea10, 2020-03-14:
The wires weren't loose wires on the serial cable and I'm still not seeing serial communications errors on the Monitor page under Communication Stats.  In the Live Register View, the Utility Voltage, Battery Voltage, Generator Time HH:MM, and Current Output E2 (EvoAC) values are changing or have changed. The value of the Hardware Version (Firmware version) has not changed.  If the false notifications are generated again, I'll change the "Additional Modbus Timeout (sec)" setting to 2.

jgyates, 2020-03-14:
The additional USB port should not be a problem (assuming they are not attempting to accessing the serial port the generator is using).

If you restart or reboot, the error count is reset to zero so it could be that you only had errors on the earlier in the week. T

The serialconfig.py output looks good. 

Based on what you are describing it sound like everything is working now, correct?  I am not sure why you had some packet loss.  If you are not seeing any errors now you may just want to set the timeout to 2 and see if that clears things up for you.

I am closing this issue for now but feel free to post to this thread if you still have an issue or if this occurs again.

closed 2020-03-14 by jgyates

mxplea10, 2020-03-14:
The error occurred again.  I changed the value of "Additional Modbus Timeout (sec)" to 2.  Just to clarify, I have never seen errors in the Monitor "Communication Stats".

[image]
[image]

jgyates, 2020-03-14:
Do you see anything other than OK in the platform stats:

        Pi CPU Frequency Throttling : OK
        Pi ARM Frequency Cap : OK
        Pi Undervoltage : OK

You could be having a power issue. Do you see the same symptom if you unplug the USB serial ports? One weakness of the pi is that if the USB draws too much current then it can lower the voltage to the other peripherals (undervoltage condition).

Before you restart, send me your logs again.

mxplea10, 2020-03-14:
I don't see any power issues.  I haven't unplugged the USB ports since combining the two applications but I will do that.  Let me know if you want to look at another log or report before I shut it down and unplug the USB cables.

[image]

jgyates, 2020-03-14:
There is nothing logged as an error since the 12th in your log files. This leads me to think that a restart of your controller may help the problem. I have seen this help with the Evolution 2.0 on a couple of other occasions when strange communications issues occurred. 

This page gives details on how to power cycle the controller:

https://github.com/jgyates/genmon/wiki/Appendix-I-Power-Cycle-the-Evolution-or-Nexus-Controller

mxplea10, 2020-03-14:
Will the controller retain it configuration and/or settings or will it have to be reconfigured?  The installer configured the generator controller and I need to document the setting if the current settings will be lost.

jgyates, 2020-03-14:
If memory serves it should retain the settings. I have a liquid cooled unit so I assume it is the same for air cooled. @liltux may be able to answer this better than I can.

mxplea10, 2020-03-14:
How can I get in touch with @liltux?

jgyates, 2020-03-14:
Since I mentioned him in this thread he will get an email.

mxplea10, 2020-03-14:
Ok, thanks.

liltux, 2020-03-17:
@mxplea10 Hello!  So a couple of things.  1. a power cycle should not remove an configurations or setups.  Upon power up the controller will always ask you the basic questions like time, date, and fuel.  It does this by default.  2.  I always recommend, if using the genmon software to not have the mobile link wifi enabled.  I have noticed on here in the past that if you use both, when generac does thier push updates wierd things happen.  
@jgyates FYI, it was a big deal at the trade show in february that generac is going to try and "lock" out third party competitors.  Just something to keep an eye on if comm settings start getting changed with firmware updates.

liltux, 2020-03-17:
@mxplea10 @jgyates I forgot to mention, 3-17 was a firmware update release.  Same firmware number 1.09, but the build number is different - 948.

mxplea10, 2020-03-18:
Hello, @liltux, thanks for the information.  I'll restart the controller and disconnect Mobile Link as soon as weather permits.