jgyates/genmon #364: comment on project, and questions

travis-farmer, 2020-04-02:
as i have not found any other way of contacting you, i have opened an issue.
i stumbled on this project a little bit ago, and i am quite amazed with the features and presentation.
I have a Honeywell 0060330 (15KW, air cooled, and i think a Nexus/Sync 1.0 controller. bought it used, and am working to get it installed myself (as of writing, just need a LP gas hookup).
i have genmon running on a Rpi 3B+, and it seems happy, though i am still waiting for the connector to actually make the RS232 connection.

now, questions:
i have ordered one of these, https://www.ebay.com/itm/USR-TCP232-302-RS232-to-TCP-IP-Module-Ethernet-Converter-Support-DHCP-Web/123767401139
so when it eventually comes from China, i can keep my Rpi inside where it is warm, and let the serial server brave the cold (Maine, USA). having never used a serial server, could you confirm it will work? i have already ordered it, but if i need a different one...

also, the wireless monitor for this generator (not the 3-light one, the one you can do minor configuration with), have you found a way to connect genmon through it? it seems to connect via the rs232 pins on the controller, so i would think the transceiver would pass the data, and with a 300m range, it would be easier to use to tap in, rather than an Ethernet cable. besides, your web interface looks far better.

jgyates, 2020-04-02:
I have not used that model, but it should work as long as it supports transparent serial (i.e. it does not impose a protocol like modbus but expects that to be handled by genmon).

This page has additional details for serial over IP:

https://github.com/jgyates/genmon/wiki/Appendix-F----Serial-over-IP

Some more info here:
https://github.com/jgyates/genmon/issues/308

I have not used the wireless monitor, but based on what I know about it it appears that that product uses the serial port so you could not run that at the same time as genmon. The wireless used is likely proprietary so that is something I do not envision undertaking.

travis-farmer, 2020-04-02:
Thank you for the information regarding the serial server i ordered. it looks like it supports transparent serial, so now i just have to wait for the slow-boat from China to deliver it.

I was going to try a pair of Xbee transceivers, but i was afraid that there may be too much data loss en-route. if i can get WiFi to reach the generator, and i think i can, then i will likely make a temporary install of the Rpi in a box behind the generator. at least while i wait for the serial server. as long as i can get the Rpi inside my house before winter comes back, or summer heat hits. fortunately i have a roll of outdoor CAT5e for the connection between the serial server and my ethernet switch.

anyway, sorry for rambling. Thanks again!

jgyates, 2020-04-02:
no problem. I am going to close this issue but feel free to post to this thread of you have related questions.

closed 2020-04-02 by jgyates