jgyates/genmon #365: Label for pi enclosure

AMTNate, 2020-04-10:
*Not an Issue* 
Has anyone created any kind of Genmon "product" label to put on your raspberry pi enclosure? I'm worried that when the service techs come to do the yearly check up on my generator that they might be a little off-put by non Generac parts inside or connected to the generator. I'm not a very creative person and I was wondering if anyone had come up with their own that they'd be willing to share.

jgyates, 2020-04-15:
I don't have one. If someone does have one I can add it to the repository.

baboss80, 2020-04-20:
@proppilot shared a label he created in a previous post here
https://github.com/jgyates/genmon/issues/31#issuecomment-346937247

jgyates, 2020-04-20:
@baboss80 

Thanks for the reminder. I added these to the repository in the Diagrams directory.

AMTNate, 2020-04-20:
> @proppilot shared a label he created in a previous post here
> [#31 (comment)](https://github.com/jgyates/genmon/issues/31#issuecomment-346937247)

I KNEW I had seen one here before! Thank you! That is so helpful.

@jgyates you can close this if you want, or you can leave it open and see if anyone else wants to share their designs.

closed 2020-05-18 by jgyates