jgyates/genmon #366: NG Fuel Consumption

swvalenti194, 2020-04-14:
Generator ran twice yesterday during outages and the fuel consumption (natural gas) for the shorter outage was higher than the longer outage...am I missing something here?  Running latest version.
[image]

jgyates, 2020-04-14:
While the duration is one factor that would determine the fuel used during and outage, the load of the generator is also a factor. The power log (/etc/genmon/kwlog.txt) contains the logged values of the generator load while the generator is running (updated every 10 seconds). This log is used to determine the estimated fuel. Each generator model has a fuel consumption algorithm that determines how much fuel is used based on how much of a load the generator is under. The higher the load, the more fuel used. I would guess that you had more energy usage in the shorter outage.

swvalenti194, 2020-04-14:
Hmmm ok just odd that one was during the day when most things were on in house and the one at night when almost most were off.  Thanks for the explanation though.

jgyates, 2020-04-14:
If you constantly see this behavior let me know (shorter outages produce larger fuel usage reports) and I can check the algorithm to make sure there is no error, however, if this is a one off occurrence then it is likely due to your electricity usage. I am going to close this issue for now. Let me know if you have any other questions.

closed 2020-04-14 by jgyates