jgyates/genmon #367: 3.3V TTL to RS-232 converter will lock-up

shpegel, 2020-04-17:
While stable once running, apparently the 3.3V TTL to RS-232 converter will lock-up if the power to it is glitched. I am using the Uxcell one from amazon in the hardware list. I have seen the lockup happen for 3 reasons: The battery is tested by the generator tech, The generator starts during a power failure, and a glitch (double tap connect) when re-powering the Genmon box. For some reason, weekly exercise starts seldom cause the problem.   To clear the lockup, just do a clean power cycle of Genmon box.

It is not convenient to go out to the generator, open it, and power cycle the genmon box.  My first thought is to use the NC contacts of a relay module and a Python script to drop the power cycle the converter but I am open to suggestions. 

Do other users have this issue?  If not, perhaps my module is the issue. Thanks.

liltux, 2020-04-17:
@shpegel That sounds like a common problem with any comms device.  Since it uses difference in voltage potential to transmit bits of data, a voltage dip or spike would cause communication issues or cause the chip to "lock up".  I would first look at the power adapter converting the 12VDC to 5VDC.  Ideally the voltage should not drop unless primary side drops below 5VDC, but that is different per each power supply/convertor.

jgyates, 2020-04-17:
@shpegel 

As a data point, I have not seen this issue before. I assume you are powering via the battery via the USB power supply listed in the project materials. You could check your voltage at the battery (or what genmon is reporting for battery voltage) and as @liltux recommends, validate you have 5V from the USB side. You could check on the Monitor page an see if you have an undervoltage has occurred.

shpegel, 2020-04-17:
Yes, I'm using the DC converter mentioned in the parts list. Battery
voltage is nominal, and I can always reach pi when converter is locked up.

liltux, 2020-04-17:
Just curious, are you powering the TTL convertor with 5 volts on VCC or 3.3 volts on VCC?

shpegel, 2020-04-17:
5v from pi.

shpegel, 2020-04-18:
I have checked all voltages and they are fine. I am thinking of trying a different converter. Here is one I found on eBay [Isolated converter](https://www.ebay.com/itm/3-3V-Single-Channel-Isolated-TTL-to-RS232-Converter-Module-SP3232EEN-115200bsp/182349564061?hash=item2a74e1889d:g:UskAAOSwK9heRSWF)

liltux, 2020-04-18:
I have been using this one : https://www.pishop.us/product/serial-hat-rs232/

shpegel, 2020-04-18:
I like that one as it will use less space in the box. Will give it a try,
thanks.

lmamakos, 2020-04-18:
I've built a prototype board that stacks on the Raspberry Pi GPIO header with an isolated RS-232 interface.  

I found an Analog Devices part (ADuM1201AR) dual-channel digital isolator that's not based on an optoisolator, so it does a great job of reproducing 9600 bps signals.  (And it should, since the specs say it has 25Mb/s bandwidth per channel.)  That, plus a Maxim MAX3221 TTL/RS-232 level converter towards the MODBUS interface should be a good solution.  I power that entire side of the serial interface from the MODBUS connector; it draws around 50mA from that power source (which is going to go up slightly when I add an power-present LED).

I also have 3 thermocouple interfaces which, along with a DS18B20, sit on a 1-wire bus connected to one of the GPIO pins.  This comes from a desire to measure oil temperature and other things that get hot inside the Generac.  This came to mind 6 hours into a power outage when I was wondering how well things were going, and was the genesis of this project of mine.

I made a stupid layout error in the prototype (and now have 19 spare tiny coasters!) and am going to spin another version with that error corrected, plus bringing out additional GPIO pins to a header (now series current-limiting resistors) with the intent of driving genmon status LEDs.  Also bringing out the I2C bus to a header (with pull-ups) for other future use.

I'm building this primarily for myself, with no other plans.  I'll probably put up the gerber files and stuff for others to get boards made if you like.  Note that I'm using 0603 sized surface mount components, and the thermocouple interface chips have an exposed pad on the bottom of the package which will require solder paste and reflow solders (either in an oven or using a hot-air rework tool).

Here's what it looks like plugged into a Raspberry Pi while I was doing some testing.  The blue wires fix the pin confusion problem on the MAX3221.

[image]

jgyates, 2020-04-18:
That sounds like a great project. I would love to have access to the gerber, etc if possible. I have been looking for an excuse to make some PCBs again. I can add it to the project repository or give you access to post it if you like.

lmamakos, 2020-04-18:
I'm finishing up the new board revision today, and  hope to send it off this weekend.  It will probably be a couple of weeks before I get the new PCB shipped to me, and I also need to order another batch of parts.. and this is the tricky bit as the thermocouple parts are out-of-stock at Mouser and Digi-Key.  I don't have a good idea of when they're going to restocked.

I do have parts to partially populate a board with just the serial interface parts, but I really do want to test the thing in its entirety before releasing the files for board to avoid a potentially broken design into circulation.

jgyates, 2020-04-20:
FYI, here is a thread on powering the converter with 3.3V vs 5V:

https://github.com/jgyates/genmon/issues/81

Since this thread I have started using 3.3V to power the converter.

jgyates, 2020-04-20:
also, this page includes a note on 3.3V usage

https://github.com/jgyates/genmon/wiki/3.1--Making-a-Cable

jgyates, 2020-04-20:
From the above link:

NOTE: If you RS-232 converter is powered using 5V the output of the receive lines to the PI are also 5V, according to the MAX3232 data sheet. The ideal solution would be to power your converter from the 3.3V pin on the GPIO header. I have yet to see a situation when powering your converter from the 5V supply did not work or damaged a pi, however the information on the PI and the MAX3232 chip does to point to the 3.3V supply being the ideal solution.

shpegel, 2020-04-20:
Thank you for the information.

I went ahead an ordered the Pi Shop RS-232 board and was wondering which
voltage they use to power the MAX232.

When I switch the converters, I will do some testing to see if I can
re-create the issue on the bench. If I can, then changing  the voltage
source for the converter would be the next test. Probably will need to
order another USB charger more closely simulate the issue, but will start
with a bench supply first.

jgyates, 2020-05-18:
I am closing this issue but feel free to post to this thread if you have any additional data to share or additional issues.

closed 2020-05-18 by jgyates