jgyates/genmon #368: [Feature Request] MQTT Topic for Start and Stop Generator

joedr, 2020-04-18:
I have been reviewing the genmqtt.py and the config/push routines seem pretty straight forward.

However, I was wondering if the Start/Stop action can be translate to a topic for MQTT.  That way, we can use any software that handles MQTT to turn on and off the generator.  This will also add a lot of automation that can be done outside the genmon project.

I will continue to look the code to understand how that on/off command gets from the webpage to the core software.

Respectfully,

joedr

jgyates, 2020-04-18:
@joedr 

that sounds like a good idea. I will try to add this soon. 

Commands are issued to genmon like this:
The web page javascript issues AJAX calls which result in code in the python flask web server being called (in genserv.py). Then genserv.py makes socket calls (via ./genmonlib/myclient.py). Then main genmon.py application fields these socket calls.

The MQTT support is just another python app that makes socket calls via ./gemonlib/myclient.py into genmon.py. The MQTT app is contained in genmqtt.py and it uses the paho-mqtt python library.

jgyates, 2020-04-18:
I have check in V1.14.04. This will add the ability to send commands. If you publish to the topic "generator/command" the payload written will be sent to genmon.

More in they commands supported are here:

https://github.com/jgyates/genmon/wiki/1----Software-Overview#genmonpy-required

More info on how genmqtt here:
https://github.com/jgyates/genmon/wiki/1----Software-Overview#genmqttpy-optional

The topic generator/command is not set by genmqtt.py, it is only read/subscribed so other apps will need to publish this value to their MQTT broker.
Let me know if you have any issues or questions.
Jason

joedr, 2020-04-18:
Hi Jason!

This is awesome.  I was able to send a payload of setremote=start but, it did not start the generator.

It seems that I am able to publish the start command but it is not been recognize by the system.  See attached screenshots.

[image]
[image]

jgyates, 2020-04-18:
Based on the screenshot it looks like your topic is:

     GEN11K/.generator/command/setremote=

and the payload is:

      start 

Instead the topic should be :

     GEN11K/.generator/command

adn the payload should be:

     setremote=start

I have not used MQTT explorer so I could be mistaken. I used node red for my testing.

You can send me your log files and I can take a look.

Another thing to look at is enabling the genmqtt debug functionality. This will put additional info in the log files. To enable debug info modify the file /etc/genmon/genmqtt.conf

       sudo nano /etc/genmon/genmqtt.conf

add this line at the bottom:

       debug = True

This will add additional info about the MQTT transactions to /var/log/genmqtt.log. Note that this will increase the writes to the SD card so I would not leave this option on as it would *eventually* cause your SD card to have a shorter life.

jgyates, 2020-04-18:
I downloaded MQTT explorer. Using that I found one issue. Not sure if this is related to your problem, but I will be posting an update soon...

jgyates, 2020-04-19:
I checked in V1.14.05. This corrected the issue I introduced in 1.14.04

Let me know if this does not solve your problem.

jgyates, 2020-04-19:
For reference:
[image]

joedr, 2020-04-19:
Jason,

This time it worked very well.  I tested and it worked but the MQTT Integration settings must have the Root Topic on blank.  See attached.  

Mine had GEN11K and that added another topic on top of "generator".

On my original configuration.... the topic was GEN11K/generator/command and the system is expecting generator/command.

Just published setremote=start over generator/command and it fired right up.  Excellent and thanks a lot!

My humble suggestion is to eliminate the Root Topic entry on the Add-Ons page or adjust it so that it can handle another layer on top of the "generator" topic.  Hope this makes sense!

Let me know if more testing is required and be more that willing to try to help you make it fail proof.
[image]

joedr, 2020-04-19:
Here is the MQTT Explorer screenshot depicting both scenarios.  The one without the GEN11K was the one that worked.

[image]

jgyates, 2020-04-19:
I checked in an update that fixed the issue. It was not recognizing the command if the Root Topic contained any uppercase letters. You should just be able to do an update from the about page, I did not change the version number.

joedr, 2020-04-19:
Hi Jason,

Just tested latest version and it works as intended.  Thank you so much for the quick update.

I think that the software is a LOT more versatile and allows for integration on different platforms.  It is in my opinion a step higher on usability.

For example, I can now setup complex routines that can range from sunset/sunrise, time, utility power status, solar system integration, complex exercise routines, turn off in case of emergency or out specs parameters.  You name it.

Just to share with the rest of the users, this is what I ended up doing:

1. Created two momentary switches: "Generator On" and "Generator Off".
2. Program two automation actions for each of the switches.
3. Generator On will publish "setremote=start" for 10 seconds and then "Generator Started" to the generator/command topic.
4. Generator Off will publish "setremote=stop" for 10 seconds and then "Generator Stopped" to the generator/command topic.

This will suffice my initial requirement of manually turning the generator On and Off from Home Assistant.  If anyone needs the code, it is listed below.

joedr, 2020-04-19:
These are the two switches using momentary integration from the Home Assistant Community

`  - platform: momentary
    name: Generator On
    mode: on
    toggle_for: 1
    cancellable: True
    
  - platform: momentary
    name: Generator Off
    mode: on
    toggle_for: 1
    cancellable: True`

joedr, 2020-04-19:
This is the automation entries.......

`- id: '123456789'
  alias: Start Generator
  description: Start Generator
  trigger:
  - entity_id: switch.momentary_generator_on
    platform: state
    to: 'on'
  condition: []
  action:
  - data:
      payload: setremote=start
      topic: generator/command
    service: mqtt.publish
  - delay: '10'
  - data:
      payload: Generator Started
      topic: generator/command
    service: mqtt.publish
- id: '987654321'
  alias: Stop Generator
  description: Stop Generator
  trigger:
  - entity_id: switch.momentary_generator_off
    platform: state
    to: 'on'
  condition: []
  action:
  - data:
      payload: setremote=stop
      topic: generator/command
    service: mqtt.publish
  - delay: '60'
  - data:
      payload: Generator Stopped
      topic: generator/command
    service: mqtt.publish`

Just remove the single quotes and remember to adjust it on the correct indentation on the File Editor inside Home Assistant.  Also, the id: is unique and the system may need to generate its own.  But, this will help understand how I did it so that it can be replicated through Home Assistant Automation Editor Page.

jgyates, 2020-04-20:
I have tagged this thread with the Monitoring Apps thread so others can find the info you posted.

closed 2020-04-20 by jgyates