jgyates/genmon #369: Most add-ons gone after update to 1.14.05

UPS99, 2020-04-19:
After upgrade to v 1.14.05, all but 3 add-ons have disappeared from the Add-ons page.  Only 3 remain - Raspberry Pi GPIO Inputs,  Raspberry Pi GPIO Outputs, and Notifications to CSV Log.  I tried 3 browsers and they all display the same way.

### Your Environment

- Generator Model: G0062440
- Generator Registers: Submit via the About page
- Genmon Version: was 1.14.03 - upgraded to 1.14.05

jgyates, 2020-04-19:
You should be able to correct this by copying the file genloader.conf from the /home/pi/genmon/conf directory to the /etc/genmon directory.

      sudo cp /home/pi/genmon/conf/genloader.conf /etc/genmon

I have seen this once before. I am guessing that either something else was slowing the system down or your SD card is failing. The latest updates do not change anything except the version number in any files except genmqtt.py. One a restart the file genloader.py is updated to contain the process IDs of the loaded genmon modules. It appears that this write took too long before the process was closed, which corrupted the file genloader.conf. Once you copy the file back over you will need to re-enable any add ons. Your other settings (any settings for each add on) should not be effected.

UPS99, 2020-04-19:
I checked the genloader.conf file in /etc/genmon and it appeared to be the same as /home/pi/genmon/conf/genloader.conf - so I just tried a reboot, and everything came back, without the need to copy the file.  I had not tried that before, not expecting it to help.  Maybe something did not fully load after the update?

jgyates, 2020-04-20:
I am not really sure what is going on here but I am glad a reboot fixed the issue. I am going to close this issue but feel fee to post to this thread if you have additional info.

closed 2020-04-20 by jgyates