jgyates/genmon #370: Genmon stopped working following annual checkup

rhorrell, 2020-04-20:
Use the template below if you have an issue or want to report a bug. If you have a question or a feature request you can ignore the questions below.

***NOTE:*** If you are having issues with your serial connection, please read this page before posting:

https://github.com/jgyates/genmon/wiki/3.6---Serial-Troubleshooting

If you are having other issues, please see the following page:

https://github.com/jgyates/genmon/wiki/3.5---General-Troubleshooting

If you need to send you logs and registers to the developer, if you email is setup and working properly you can click send your logs on the About page in the web interface.

### Expected Behavior

{Please write here}

### Actual Behavior
Genmon stopped reporting (Not receiving data)

Sample from /var/log/mymodbus.log
2020-04-20 16:17:50,226 : Error: timeout receiving slave packet for register 05ee Buffer:0
2020-04-20 16:17:53,257 : Error: timeout receiving slave packet for register 0235 Buffer:0
2020-04-20 16:17:56,296 : Error: timeout receiving slave packet for register 023b Buffer:0
2020-04-20 16:17:59,330 : Error: timeout receiving slave packet for register 023a Buffer:0
2020-04-20 16:18:02,364 : Error: timeout receiving slave packet for register 003a Buffer:0
2020-04-20 16:18:05,399 : Error: timeout receiving slave packet for register 003c Buffer:0
2020-04-20 16:18:08,435 : Error: timeout receiving slave packet for register 023d Buffer:0
2020-04-20 16:18:11,463 : Error: timeout receiving slave packet for register 05fa Buffer:0

### Steps to Reproduce (including precondition)

Genmon has been working perfectly for the past 3 years THANK YOU!!!!

Electrician came by today to due annual maintenance and the Genmon stopped working.
He swears he didn't do anything to the controller (firmware updates, etc).

He did say that newer firmware would disable 3rd party monitors.

### Screenshot or Pictures relating to the problem (if possible)

{Please write here}

### Your Environment
 Raspberry Pi Zero W

- Generator Model: Generac G0070431
- Generator Registers: Submit via the About page
- Genmon Version:  V1.14.05

jgyates, 2020-04-20:
Have you checked the cable to make sure that it is connected. It may have a marginal connection. 

I would try a loopback as described here:

https://github.com/jgyates/genmon/wiki/3.6---Serial-Troubleshooting

This will make sure the pi and RS-232 converter are still working.

If they are and you still can not received any packets then I would try to power cycle the controller as described here:

https://github.com/jgyates/genmon/wiki/Appendix-I-Power-Cycle-the-Evolution-or-Nexus-Controller

Let me know how it goes..

jgyates, 2020-04-20:
I got your logs. It does appear to be a serial comm issue.

rhorrell, 2020-04-21:
Looks like the service technician may have damaged the cable.  Thanks for your help.

jgyates, 2020-04-21:
No problem. I am closing this for now but feel free to post to this thread if you have other questions on this topic.

closed 2020-04-21 by jgyates

rhorrell, 2020-04-23:
Made a new serial cable and reinstalled unit.  Working great again, Thank you!!!!