jgyates/genmon #373: How to supply temperature data to genmon for display or other usage

lmamakos, 2020-05-07:
I've been working on building a board for a Raspberry Pi that contains:

- an electrically isolated RS-232 interface
- 3 thermocouple interfaces for measuring temperatures inside the generator
- on-board DS18B20 temperature sensor + screw terminals for additional external temperature sensors
- various expansion connectors that expose an I2C interface and 6 current-limited GPIO connections for driving LEDs

[image]

I have the hardware design complete, and after spinning the second spin of the board, I have only a few remaining regrets (but no more blue bodge wires! :-)  See https://github.com/lmamakos/RPi-pHat-Thermocouple if you're curious.

While the hardware is complete, there's still some software yet to do such that this is useful for genmon.  Now, minimally I expected to hack up a python script to read the thermocouple temperatures and the other 1-wire bus temperature sensors and publish these to MQTT.  My own environment is using Home Assistant and I already integrate lots of stuff with MQTT.

My question is if there's interest in having genmon access this somewhat more directly so it could display them within its web UI.  I'm not sure if there's a common pattern for conveying this sort of data to genmon, and what, if anything, additional I should implement.

On the one hand, MQTT is a really handy integration mechanism and if you don't have an MQTT broker, you could just spin up mosquitto locally for the purpose of connecting genmon...? 

And advice here would be useful.

jgyates, 2020-05-07:
@lmamakos 

The quickest way to get the data displayed is to store the data in JSON object format in a file in the /home/pi/genmon directory with the following name:

        /home/pi/genmon/userdefined.json

The file would look like something like this but this is just and example that I use, it couod be as simple or complex as you need, but just needs to be a JSON object:

    {
        "Pulse Count": {
            "Value": "19981670", 
            "Time": "Thursday May 7, 2020 13:28:54"
        }, 
        "Power": {
            "High": "22990", 
            "Low": "0", 
            "Value": "10158", 
            "Time": "Thursday May 7, 2020 13:28:54"
        }, 
        "Temperature Node": {
            "Temperature": "73.4", 
            "Battery": "2.6", 
            "Humidity": "35.2", 
            "Pulsecount": "1", 
            "External Temperature": "73.4", 
            "Time": "Thursday May 7, 2020 13:28:17", 
            "Rssi": "-65"
        }, 
        "Solar": {
            "Load": 12620.0, 
            "PV": 19980.0, 
            "Grid": 7360.0, 
            "Time": "Thursday May 7, 2020 13:24:52"
        }
    }

once you write this file, genmon will display it on the Monitor page like this:

[image]

Alternatively, I could possibly add an option to put this data on the status page, but I would need to know the data format or have a unit to test with.

lmamakos, 2020-05-07:
That seems like an easy solution.  Is the pathname for that file a configuration option?  I think I'd probably want that in /tmp or some file system mounted on a RAM disk to avoid needless extra write cycles to the microSD card.  Or maybe just a symlink..

jgyates, 2020-05-07:
The path is not currently an option. I will add that to the feature TODO list. You could use a symlink to a RAM disk.

jgyates, 2020-05-07:
Just to clarify, the path is the path that genmon is installed so if you opted to install in a directory other than /home/pi/genmon then that should still work.

    <path that genmon is installed>/userdefined.json

griffths, 2020-05-18:
@lmamakos 

I am interested in adding this to my current setup and consider myself a reasonably competent cook and can follow recopies but in no way consider myself a chef.....

I see the supplied BOM but not a specific link to the PCB.   You mention using JCL PCB but it is unclear from their website if they also do the wave soldering.  I can manage through hole work but have no way to do surface mount stuff.

If JCL can do everything, do I need to supply them with the parts called out in your BOM or do they handle everything?

Regards,

Jay

jgyates, 2020-05-18:
I have updated genmon to V1.14.07 which includes the ability to have a user defined path for the userdefined.json file (see advanced settings by double clicking the gear icon in the upper right of the web interface).

lmamakos, 2020-05-18:
JLC doesn't have the ability for me to store a PCB design that you can just reference.  If you want a board, you'd have to upload the gerber files and have them construct 5 for you.

JLC does do board assembly of parts.  However, they require a BOM listing for _their_ parts; you can't ship your parts to them.  Further, they do not stock the thermcouple devices.  I did a quick check, and I think they have the parts for doing the RS-232/TTL serial interface... HOWEVER, I've not tried to do this, and I think in one case they have the part, but it might be in a different sized package.

If you were mostly interested in the RS-232/TTL interface converter, it's probably possible to hand solder all those parts on the board.  The devices are quite small; the passives are mostly all 0603 -- so if you go down that path, make sure you order extras.  If they flick out of your tweezers on the floor, they're just gone.  They disappear into a parallel dimension to join the socks you lose in the clothes drier.  The impediment to hand assembly is that the PCB pads are small and there's really nothing exposed to put a soldering iron on; the pads are intended for solder paste and reflow, either in an oven or using a hot-air tool.  

The thermocouple devices are essentially impossible to deal with unless you use solder paste and some sort of reflow.  There's a pad on the bottom of the TDFN package, and really no exposed leads on the side of the package. 

That all being said, this is the first board I've done using paste and hot air reflow and it went better than I expected.  For me biggest challenge was just being able to see all the parts and using a binocular microscope made it much easier.  You can watch me assemble the second version of the board here: https://www.youtube.com/watch?v=6pSCB2km2_s speeded up quite a bit.  It's much more entertaining with the Benny Hill theme played behind it, but YouTube is unhappy about the audio track :-)

Something I've learned in the last week is that the MAX31855K thermocouple devices do not like thermocouple sensors where the hot junction is grounded.  They will register a fault when you try to take a measurement.  So I cannot directly use the really cheap type-K thermocouples from Amazon and other sources without carefully thinking about how they will make contact with stuff inside the generator enclosure.  I'm thinking of forming a small bead of high-temperature epoxy around the exposed junction as insulation.

Anyway, I was thinking that it might be possible to respin the board with only the RS-232 part of the circuitry and using the parts that JCLPCB stocks so that there's an easy way to order something that has all the surface mount parts installed.  (You'd still have to solder on the connectors - they don't do through-hole assembly as part of that service.)

I'd hold off a week or so before doing anything.  I have some updates to the README file in the repository that's not been pushed to github yet that might be helpful.

griffths, 2020-05-18:
Thanks for the quick update and YouTube video link.

I am not in a big hurry so can wait if you feel there is a possibility of another revision of the hardware portion.

I am primarily attracted to the temperature reading portion of your project as i have a liquid cooled Generac and would like to know when/if the battery warmers and block heater are active during the winter as well as ambient temperature within the generator enclosure

If I have interpreted your description properly there is an onboard thermocouple sensor as well as connections for three external sensors on your project.

That said, I also like the integration of multiple functions on your board to clean up and simplify the wiring of the hardware within the enclosure.

dennis48755, 2020-05-18:
Hi Louis,

What hot air station are you using/recommend? I am in the market.

lmamakos, 2020-05-19:
@griffths the reason for the thermocouple devices was to measure relative high temperatures, like the oil temperature, which I imagine measuring via the cooler on the side of the motor.  Heck, with a thermocouple, I could measure the temperature of the exhaust manifold if I could figure out a way to mechanically attach to it!   If you want to measure something that **never** exceeds about 125C, then you can use a "1-wire" temperature sensor that's in a plastic package.  This is exactly the on-board sensor in that 3 pin package with the leads sticking out the bottom.  You can get these DS18B20 devices for about $4 each, or for just few dollars more, packaged on the end of a wire in a metal probe arrangement, allegedly watertight.

Here's a photo of one of those probes; since these are individually addressable, you could have more than one of these sensors wired in parallel with the on-board sensor (and also the three thermocouple interfaces which use the same shared bus protocol).  Here you can see 3 of the thermocouples with their exposed junctions, and the DS18B20 on the end of that black cable with the silver-colored metal cap:

[image]

That 3 pin terminal block was intended to connect to external 1-wire bus sensors like these.  For example, these devices: https://www.amazon.com/DS18B20-Temperature-Waterproof-Digital-Thermal/dp/B082HJB7C7 picked at random; there's a bunch of suppliers on Amazon, ebay, aliexpress, banggood, etc.

So minimally, you could try to build the board and leave off most of the parts associated with the thermocouple interfaces.  Probably even hand solder the parts if you're careful and have some magnification so you can see WTF you're trying to do :-)   I'm "older" and my eyesight isn't as good as it used to be, so YMMV.

@dennis48755 - there's a bunch of inexpensive hot-air tools around.  The unit that I bought is an X-TRONIC 4000-something.  I wouldn't recommend it specifically.  It's a combo device with both the hot air tool and a soldering iron.  While the soldering iron is actually fairly nice, time has marched on and I just don't use that soldering iron any longer.  I much prefer the Hakko clones that use the T-12 cartridges which have much better temperature control, heat up much faster and are just a pleasure to use.  I'm using a KSGER soldering iron, and I bought a handful of different cartridges for not much $$$ at all - these have the heating element, temperature sensor and actual tip all integrated and close together.  Lots of YouTube videos reviewing these.

jgyates, 2020-07-13:
This board is supported via this Add On to genmon:

https://github.com/jgyates/genmon/wiki/1----Software-Overview#gentemppy-optional

lmamakos, 2020-07-13:
Thanks @jgyates for adding support!  Note that the work what was done will support Maxim DS18B20 temperature sensors that you can connect to a GPIO pin on the Raspberry Pi (with a pull-up resistor; I used a 4.7K in my design).  You don't need my board with the thermocouple interfaces and isolated serial interface if you just want to measure some temperatures.

jgyates, 2020-09-29:
I am closing this issue thread as the wiki has the related info. I will also put a link to this thread from the wiki.

closed 2020-09-29 by jgyates