jgyates/genmon #374: Generac will begin encrypting evolution 2.0 controllers with further updates. Will this block genmon use

grsthegreat, 2020-05-08:
Use the template below if you have an issue or want to report a bug. If you have a question or a feature request you can ignore the questions below.

***NOTE:*** If you are having issues with your serial connection, please read this page before posting:

https://github.com/jgyates/genmon/wiki/3.6---Serial-Troubleshooting

If you are having other issues, please see the following page:

https://github.com/jgyates/genmon/wiki/3.5---General-Troubleshooting

If you need to send you logs and registers to the developer, if you email is setup and working properly you can click send your logs on the About page in the web interface.

### Expected Behavior

{Please write here}

### Actual Behavior

{Please write here}

### Steps to Reproduce (including precondition)

{Please write here}

### Screenshot or Pictures relating to the problem (if possible)

{Please write here}

### Your Environment

- Generator Model: {Please write here}
- Generator Registers: Submit via the About page
- Genmon Version: {Please write here}

liltux, 2020-05-08:
I don't believe you can encrypt modbus rtu

grsthegreat, 2020-05-08:
i dont know. according to the email i received, there doing this to stop 3rd party access to controller information. it will also prevent the use of Generacs own wireless monitor systems and only allow use of their proprietary wifi and mobile link systems. it wont effect evolution 1.0 controllers, as they need usb assisted updates, but i wont be updating my 1.0 system any more.

liltux, 2020-05-08:
Was this an email from Generac?

grsthegreat, 2020-05-08:
yes. as a heads up. man, i wish i didnt accidentally delete it.

grsthegreat, 2020-05-11:
your post had me go look up the post recap from the march webinar on dealer connection...lo and behold the mention of port encryption IS noted there. it definitely states there that 3rd party connectivity solutions will NO LONGER be able to have automatic access to receive information and controlling generator. they will be introducing port encryption. this is to start 1st quarter 2020. The email told me that they will push this update to installed 2.0 controllers also. 2.0 evolution generators that are updated will encrypt the ports. so my customers that are using Genmon are probably going to be screwed.

personally, i will not be updating any controller that has version 1.0 evolution AND Genmon (including my own).

liltux, 2020-05-11:
@grsthegreat sounds challenging, was there mention of the liquid cooled generators?

swvalenti194, 2020-05-11:
Does this mean it'll automatically get pushed to the controllers or it would have to be manually done?  Anyway to block this if it's automatically done OTA?  Dumb question but how could they push OTA as I don't think my generator has any comms with Generac?

rnatalie, 2020-05-11:
It’s not even possible on the old Evolution 2.0.   I don’t know if the Sync 3.0 version will autoupdate or not.

> On May 11, 2020, at 10:29 AM, swvalenti194 <notifications@github.com> wrote:
> 
> 
> Does this mean it'll automatically get pushed to the controllers or it would have to be manually done? Anyway to block this if it's automatically done OTA? Dumb question but how could they push OTA as I don't think my generator has any comms with Generac?
>

grsthegreat, 2020-05-11:
all i know is the evolution 2.0 are wifi accessible, and currently use the wifi to do the forced updates. with the V1.0 i had to upload all updates to a flash drive and install them when i serviced generators. i do not have to do this to the V2.0 units as its done thru wifi. 

now if the generator isnt connected to the wifi at home, i guess the update wont take effect. the only problem is there are updates that are critical for these units. Generac has pushed thru several critical updates over the years that affect  generator operations and when you do updated they include all changes as you cant pick and choose. luckily the V1.0 probably wont see many needed updates in the future so my unit is probably safe, but  im worried about some of my clients that i talked into getting Genmon units instead of generacs mobile link setup.

the last liquid cooled unit i installed last year had an evolution controller but was not wifi capablle, so i would assume it will not be affected. older liquid cooled units that i was installing had versions of the Nexus controllers so wont be affected.

zekyl314, 2020-05-11:
As a home owner, do I have the ability to get these 1.0 updates myself?

grsthegreat, 2020-05-11:
> 
> 
> As a home owner, do I have the ability to get these 1.0 updates myself?

no, they are accessed thru Generacs dealer service portal. they have to be installed thru dealer portal on the generator. if you use a dealer for service/valve adjust, etc they are supposed to provide the updates. 

i have seen them sold on flash drives thru ebay in the past... not sure if they still are. 

current evolution 1 firmware is V1.21

liltux, 2020-05-11:
@everyone:
Evo/Evo1.0 or Sync 2.0 are the same controller.  These controllers do not have WiFi capabilities and firmware must be updated manually via USB.  Homeowners need to contact their local dealers for firmware.  It does not require a dealer access to update firmware.
Evo2.0 or Sync 3.0 are the same controller. These controllers do have WiFi capabilities.  The WiFi can be turned off.  These controllers are firmware updatable via WiFi push OTA and via USB.
EVO 1 type controllers only have, to my knowledge, RS232 capabilities.
EVO 2 type controllers have a RS232 and RS485 Port.

grsthegreat, 2020-05-11:
you know, your right. i have always installed updates thru dealer input area, but now that i think about it they can also be updated thru the standard edit screen where you change time/date. i just never access it thru that area because i do my updates while accessing dealer screen and resetting maintenance push notices screen.

jgyates, 2020-05-11:
Just curious. Is there a way to downgrade a Evo 2.0 controller's firmware from the console via USB? I assume Evo 2.0 still has USB firmware update capabilities? If yes, then people who have auto-upgraded 2.0 controllers could downgrade to keep using genmon, assuming they could get their hands on older firmware.

grsthegreat, 2020-05-11:
Not to my knowledge. When you install usb drive the generator checks to see if the installed firmware is older or not. Ive never seen where you can downgrade it. I once had a unit tell me the firmware was the same as existing firmware and all it prompted was exit.

buzzshot, 2020-05-12:
I haven’t seen what the update software looks like but if it’s not an exe I am sure you could mod the old version to look like the new version. And then dump that to the generac. That is if some one has an old version copy kicking around

liltux, 2020-05-12:
@jgyates if you have an old revision I have been able to load an older firmware into controller.  I have done this on both Evo1 and Evo 2.  It does require wifi disabled on the Evo 2.  The Evo 2 will compare flashdrive to remote server and only install the latest version.

hokie21, 2020-05-13:
I would have thought that selecting Modbus rather than a proprietary protocol would have been because Generac wanted their products to integrate easily with 3rd party monitoring and control systems.  There are other monitoring solutions in addition to Genmon that this change will break. It looks like https://www.omnimetrix.net/ is also using the Modbus connection to Generac generators. https://www.omnimetrix.net/wp-content/uploads/2018/05/IM-706-TrueGuard-PRO-NEXUS-Panel-Installation-Rev-F.pdf. 

Hopefully Generac dealers as well as the folks at Omnimetrix are making some noise about this with Generac.

lakee911, 2020-05-15:
It's quite possible that their "encryption" is something like a shift cipher or simply bastardized Modbus. I bet that it won't be tough to crack.

UPS99, 2020-05-16:
There can't be much extra storage, CPU power, or free memory to do any serious encryption.  They could, of course, copyright some part of the code, making it more difficult to distribute it.

skipfire, 2020-05-16:
Is it known what firmware version will introduce this?  I just disabled the WiFi on mine, but no clue if I have the encryption version yet and I have all the parts on the way to hopefully get this going.

grsthegreat, 2020-05-17:
No, but the email  i received said its coming in a future update soon.

lakee911, 2020-05-17:
What's the current version? Presumably, the next version will introduce it.
If nothing else, the current version is the last compatible version until
we know better.

liltux, 2020-05-17:
Evo 1 is 1.21
Evo 2 is 1.9
Note they can change the firmware and not the revision number(if they so chose).  The most recent update to Evo 2 was 1.9 (build ###) but it still only shows up as 1.9.

UPS99, 2020-05-19:
Here is more info about the encryption:

https://fleetsupport.zendesk.com/hc/en-us/articles/360042081271-Controller-Encryption-FAQ-s

hokie21, 2020-05-20:
Interesting that they only care about fixing the "security" problem on air cooled and not liquid cooled generators. @jgyates can you speak to Generac and see if they might consider "authorizing" Genmon? I'd be curious what kind of response you would get from them on this.

jgyates, 2020-05-20:
I would be happy to reach out to them if I had a contact. Other than going to their web site and filling out the contact us form, I don't have any contacts with Generac. My guess is that they will ignore me. The only reason I can see for this encryption is to drive people to use their service.

lakee911, 2020-05-20:
Send me your email address, Jason, and I'll send you the contact that I
have. Maybe he'll be able to get you pointed in the right direction...

jgyates, 2020-05-20:
you can send me email at generatormonitor.software@gmail.com

lakee911, 2020-05-20:
Jason,

   This is the only contact that I have at Generac. He MIGHT be able to get
you pointed in the right direction. He helped me w/ my initial install when
I needed to do two-wire start and even sent some free parts
(pre-terminated pins and wires). I recently contacted him for help
regarding my fried controller and he gave me a short response to simply buy
from Ziller. I think it was my local rep, for work, that had originally put
him in touch with me, but that guy at my local dealer has since moved on.
Anyhow, it's probably a safe bet that this guy would not know me.

Brian.Baughman@generac.com
Brian Baughman

Generac Power Systems, Inc.

Technical Applications, Building Codes

Office 262-544-4811 x2046

Cell 262-290-1652

   Good luck!

Jason

Spuds, 2020-05-20:
Received a message last night from Mobile Link

> Your generator has successfully uploaded a system upgrade. Your generator is ready to run with the latest software updates.

This update seems to have broken the communications.  The generator now shows firmware 1.10.59 and the genmon status just shows Modbus Exceptions : xxxxx  which is continually increasing.

liltux, 2020-05-20:
Well it was fun while it lasted....

UPS99, 2020-05-20:
The encryption argument by Generac is not valid.  No one would have access to hack modbus.  It they legitimately felt the need for added security - encrypting the ethernet / Wi-Fi traffic would be the only effective place to do it.  This is very likely related to their new warranty requirements, which only provide the full extended coverage to customers who purchase and pay for the MobileLink plan and keep their generator connected to Wi-Fi.

skipfire, 2020-05-20:
I actually came here because the first year of my MobileLink ended and I hated how little info it provided.  Got the parts today, this weekend I'm building the cable and hopefully getting this all working on a Pi Zero (hopefully).

grsthegreat, 2020-05-20:
just make sure you dont hook the generator up to wifi if its an evolution 2.0 controller. if its the older evolution 1.0, dont do any more firmware updates.

i just set aside one of my new evolution 1.0 spare controllers(purchased last year) onto my personal shelf in the shop. now i wont have to worry that a replacement V1.0 controller will be encrypted and wreck havoc on the genmon unit in my own 22kw unit.

skipfire, 2020-05-20:
I disconnected it from WiFi as soon as I saw the post, so hopefully I got it before the update went to it.

jgyates, 2020-05-20:
@skipfire if you are performing a new install, don't use V1.14.09 to do the install. I checked that in last night, but I found a bug that has been fixed in V1.14.10 (checked in a few min ago). If you did use V1.14.09 let me know and I can tell you how to fix it. The bug wires an invalid command line option to the crontab on a new install.

skipfire, 2020-05-20:
Thanks, I haven't downloaded it yet, hoping to chew through the whole process including cable on Saturday (if the weather is ok).  I also still have to take apart the generator enclosure to find the plug.

ebovine, 2020-05-23:
I'm confirmed as having this problem as well. Unfortunately I didn't see the threads prior to them pushing the update.

mhump711, 2020-05-25:
> Received a message last night from Mobile Link
>
>
> This update seems to have broken the communications. The generator now shows firmware 1.10.59 and the genmon status just shows Modbus Exceptions : [...]

Can confirm mine stopped working and I am on 1.10.59 :(
Sad they went through all this to make a quick buck.

grsthegreat, 2020-05-28:
Well it’s official... i am no longer installing Generac generators. I had a talk with  my dealer rep and let them have it with what i thought about their money grab with encryption. I am no longer going to bid installing generac units. If i decide to Keep installing units, im going to switch to Kohler units. I have 6 customers that want generators, and im going to have to  contact them And cancell The planned installs... no signed contracts yet, but was going to sign next week.

I may just stop i stalling generators all together.

I am still going to service Generac, but only for my existing clients. Im not taking on new Generac customers.

jgyates, 2020-05-28:
If anyone with a Kohler unit was to work with me I can add support to the project.

hokie21, 2020-05-28:
If you want to register your displeasure with Generac, here is the contact form:
https://www.generac.com/service-support/contact-us/online-form

ebovine, 2020-05-28:
Generac will likely use security as a justification for this. As one in the industry with ICS experience, their actions address an incredibly low risk item and are aimed at securing subscription revenue.  Promotion of kohler over Generac appears to be a good strategy going forward.  My installer is already tired of dealing with Generac’s certification game.

jonathanpisarczyk, 2020-05-28:
I have lost access to my generac.  So annoying! We should start a change.org ?

hokie21, 2020-05-28:
I think a complaint to Generac directly using the online contact form I linked above will probably be more effective.  I suspect that unless the Generac dealers revolt and stop specifying Generac for new installs, Generac has their minds made up on this.

liltux, 2020-05-28:
Contact Generac directly and they may unlock your controller?  One offs I doubt will cause Generac problems, but they may force you to void your warranty to use third party product.

gzebrick, 2020-05-28:
I wonder if the Honeywell-branded version of the Generacs will also have the same encryption? (I have a Costco/Honeywell/Generac 17KW with an earlier Evolution-1 so as long as I don't manually update it I think I'll be safe)

UPS99, 2020-05-28:
"Generac will likely use security as a justification for this. As one in the industry with ICS experience, their actions address an incredibly low risk item and are aimed at securing subscription revenue. Promotion of Kohler over Generac appears to be a good strategy going forward."

Encrypting Modbus is absurd.  If this was really about security, they might add encryption to the Wi-Fi traffic. 

Generac has targeted their own customers as the enemy.  In particular business users of the expensive large scale monitoring solutions OmniMetrics and Power Telematics.  Complaints may fall on deaf ears.  Most effective might be if a large stockholder heard the details...

jonathanpisarczyk, 2020-05-28:
How about a class action lawsuit?

hokie21, 2020-05-28:
I think that industrial customers who use SCADA interfaced to Modbus are primarily using liquid cooled generators. That's the case with my company. That is likely why liquid cooled isn't included in the announcement.

I think a class action suit would likely result in a settlement with some money for the attorney and we would get free MobileLink service for 6 months. We probably wouldn't get the outcome we desire. If Generac is gong to change their approach, it will need to come from pushback by the dealers.

grsthegreat, 2020-05-28:
Liquid cooled as of right now do not have wifi ability, thats why they wernt targeted. As soon as they add wifi to those, they will be encrypted. 

Honeywell units have same controllers as generacs guardian lines. The evolution 1 units will require a manual update. Ill be generac will add this to future evol1 updates. Thats why i will no longer update my own personal evol 1 controller. If a client wants an update, sure ill oblige.

hokie21, 2020-05-28:
Lots of critical infrastructure is backed up by generators. These customers for the most part do not want and will not allow Internet connections to their generators due to their information security policies. They connect their generators to their own network monitoring centers over private networks. I would be surprised if Generac decides to alienate these industrial and critical infrastructure customers.

liltux, 2020-05-28:
Generac is moving Mobile Link into the larger liquid cooled, industrial units later this year.  They are just wanting to get into the monitoring market, being their units they can modify as that want to.

gzebrick, 2020-05-28:
I wonder how hard it would be to just replace the controller completely in the smaller air-cooled generators. I'd have to see what the I/O list looks like but low cost single-piece programmable controllers (with displays, built-in web servers, data ports, high resolution analog and digital inputs and outputs) are pretty common now. It's not a nuclear reactor, so a full speed PLC may not be required. In fact, the Pi probably has more horse power than the Evolution controller built into  my 17Kw. 

That might be the next step - after-market full replacement controllers with integral data ports (232/485/IP) with web servers (for use after the warranty has expired). - and they's speak Modbus / BACnet / and other protocols as needed.

UPS99, 2020-05-29:
Omnimetrix may already have a work-around.  This install PDF for Generacs, dated Tuesday, 5/26, refers to encrypted and non-encrypted Modbus and has an alternate direct wiring diagram and pictures to bypass the plug-in interface:

https://www.omnimetrix.net/wp-content/uploads/2020/05/IM-1087-01-TG2-Nexus-EVO_4-Alarm-Modbus-Remote-Start-Install-Guide.pdf

hokie21, 2020-05-29:
The work around leaves much to be desired and has no visibility to the detailed error information on the bus. Before I had Genmon I used the transfer indication and the relay closure output connected to my house alarm system. It looks like they are doing something similar to this.

UPS99, 2020-05-29:
I don't have info on the capabilities of various Omnimetrix models.  However the brand was specifically mentioned in Generac info as one that would be disabled by the new firmware.  That doc may cover only an add-on device.  All of their documentation may not be on line, but still, this might suggest a starting point, perhaps by direct connection to the evo2 board.

ebovine, 2020-05-29:
> I wonder how hard it would be to just replace the controller completely in the smaller air-cooled generators. 

This was my first thought as well so I opened the controller that evening with the intent to begin an open controller project.  It’s not too complicated, but certainly not a near term solution.

grsthegreat, 2020-05-29:
I think a evolution 1 controller would also work in a evol 2 unit, lose the wifi cable.....i wish i could try it.

[93 more replies]

closed 2021-04-06 by jgyates