jgyates/genmon #375: Info on how to get and add free LetsEncrypt signed certificates to genmon..

rnatalie, 2020-05-09:
wrote a brief addition to the WIKI to explain how to install free LetsEncrypt certificates into GENMON so chrome and other browsers stop griping about the self-signed certificate.

Alas github doesn't handle forks for the wiki like it does the sources.   Here's the link into my fork

https://github.com/rnatalie/genmon/wiki/Appendix-K----Adding-LetsEncrypt-certificates-to-Genmon

I'm thinking of fixing the genmon code itself to handle the "webroot" authentication so that the update of the certifciate can be automatic.

jgyates, 2020-05-09:
I can add you to the project if you like. Just make the webroot authentication and optional feature. Let me know if you have any questions if you decide to jump in.

jgyates, 2020-05-09:
I added your wiki entry to the project:

https://github.com/jgyates/genmon/wiki/Appendix-K-Adding-LetsEncrypt-certificates-to-Genmon

hokie21, 2020-05-13:
@rnatalie Can the certificate renewal step be automated with cron so that it is done automatically just shy of every three months?

rnatalie, 2020-05-14:
I’m working on that!  

 

@rnatalie <https://github.com/rnatalie>  Can the certificate renewal step be automated with cron so that it is done automatically just shy of every three months?

jgyates, 2020-07-11:
I am closing this for now. Feel free to post to this thread on related issues.

closed 2020-07-11 by jgyates