jgyates/genmon #376: Notification when comm fails

jwtaylor310, 2020-05-18:
My copy of Genmon continues to run beautifully -- thank you for a great product!

Is there any way that Genmon can be configured to send an email alert when communication fails?  I have my Raspberry Pi connected to the Evolution 2.0's serial port.  Any time the Pi reboots or there is any other interruption of communications between the Pi and the generator, the generator apparently closes its serial port and I have to cycle the controller's power to get communications restored. (This is a known problem documented elsewhere on this site).  My concern is that I don't always know that this has happened unless I check Genmon's status page.  I will then see zero values for Battery Voltage and Utility Voltage and when I open the Monitor page it will show large numbers of timeout errors and the packet counts don't increment.  So, can Genmon send an alert when no packets have been received for a period of time or some other monitored item (like battery voltage) goes out of range due to the lack of communication?

jgyates, 2020-05-18:
The software does report communication failures by email, however with the use case you describe I can see and issue. This is the way it currently works:

Once the software is loaded and communicating it read thru all the modbus registers, once the initial first read of each register is complete then the software attempts to identify the controller. Once this is complete then the software sets an internal flag that notes the initialization is complete.  Once initialization is complete, if a modbus transaction is not completed every 2 seconds (8 seconds for modbus over TCP) then an email is sent stating there is a communication problem.

The software does not send communication failure notifications if the initialization was not complete. The idea was to put a check for a bad connection or hardware, but not to send warning messages if you are just bringing up the system without being connected.

In your case, if the pi is rebooted and serial comms are never working from boot, then you will not get a notice. 

Let me think about how best to implement this and see if I can make this work for you.

jgyates, 2020-05-19:
I checked in V1.14.08. The way this works now is this:

The code that checks for communication failures now waits for the initialization to complete for no longer than 30 seconds, then it performs the normal communications check described above. Let me know if you have any questions or if you have any issues.

jwtaylor310, 2020-05-19:
Thank you!  I have installed V1.14.08 and will let you know if it works as planned next time my system reboots.

jwtaylor310, 2020-05-19:
When I installed V1.14.08 I received six "Generator Monitor Starting" emails over a period of 2 minutes followed by a single notice that the system is in "Auto, Off-Ready" mode.  Presumably there should only be one "Starting" notice.  I don't recall whether this was the case in previous versions.  Not a big deal, but thought you should be aware.

jgyates, 2020-05-19:
That is likely due to the flask library not restarting. The software relies on flask to run the web server. If a failure is detected it attempts to restart again (hence the multiple emails). A reboot should fix that problem from occurring. The flask library is and external library that supports web applications in python (https://flask.palletsprojects.com/en/1.1.x/)

jgyates, 2020-05-19:
I am going to close this thread for now but feel free to post here if you have questions or issues on this topic.

closed 2020-05-19 by jgyates