jgyates/genmon #381: Modbus exceptions after installing V1.14.09

jwtaylor310, 2020-05-20:
I installed V1.14.09 this morning and all communications are now being reported as "Modbus Exceptions" - otherwise comms show no errors.  Generac pushed a firmware update V1.10 to my generator yesterday afternoon.  Is my problem the result of the Genmon update or is this the result of encryption being added to the controller (Evolution 2.0)?

grsthegreat, 2020-05-20:
Weird. On generace service site they dont list V1.14 so i dont know what was included. They did warn about encryption in future updates. This sucks.

I am not sure how to roll it back

Spuds, 2020-05-20:
Same issue here, received a message last night from Mobile Link that said

> Your generator has successfully uploaded a system upgrade. Your generator is ready to run with the latest software updates.

This update seems to have broken the communications. The generator now shows firmware 1.10.59 and the genmon status now just shows Modbus Exceptions : xxxxx which is continually increasing.  Tried the latest genmon 1.14.09 with same results.

jwtaylor310, 2020-05-20:
I sure hope that all of your work on Genmon won’t go to waste due to an encryption issue.  In my view that would be very short-sighted on the part of Generac.  --John

jgyates, 2020-05-20:
Can one of you send your logs on via the About page?

Spuds, 2020-05-20:
I've sent the logs, hope they go through

jwtaylor310, 2020-05-20:
I have sent you a copy of my logs….

jgyates, 2020-05-20:
@jwtaylor310, I got yours.
@Spuds, I did not get yours.

There are several types of mobus exceptions. The exception that is used here, according to the logs is 5 = Acknowledge. Which is the modbus way to say, I got your packet, but I am not responding with data.

I have a few things to try, let me get my thoughts together on an approach. I will keep you updated.

Spuds, 2020-05-20:
I tried again .... FWIW my file mymodbus.log has lots of the same

> 2020-05-20 14:50:11,653 : Modbus Exception: Acknowledge: 05 , Modbus Command: 83
2020-05-20 14:50:11,655 : Error Receiving slave packet for register xxxx

jgyates, 2020-05-20:
Yes, that is the exact message in the other log files.

I have a few test to run. I have someone with an Evo2 that is helping test. We are waiting on some hardware to perform the test so it may be a bit before we know the results.

jwtaylor310, 2020-05-20:
If it would help I can capture the packets on the serial port to/from my Pi and dump them into a file for your review.  Let me know if that would help.  --John

jgyates, 2020-05-20:
Thanks for the offer. Based on the logs, I think I have a pretty good idea of what is happening on the Rs-232 serial interface. What I need is to get a look at is what is happening on the RS-485 serial port (the one that is connected to the wifi module). I have someone who already volunteered to do this (just waiting on RS-485 adapter). If the wifi module is using modus  (over RS-485) as well, then we can see if it doing any special sequence of data to enable responses, or possibly start using the modbus or some other protocol over RS-485 on Evo2. At least that is the idea. We will know more once we get a look at the data.

jwtaylor310, 2020-05-20:
OK.  Good luck with the RS-485 interface!  --John

skipfire, 2020-05-22:
How can I find what firmware I have?  The firmware path in the Evolution 2 controller just goes to update without showing the current version, and when I plugged in, GenMon shows CRC errors, but I don't know if that is from an error with wiring or because I got the updated software.  Any suggestions?  I just tried sending the registers and logs, just in case.

jgyates, 2020-05-22:
From the top most menu enter this sequence : Up, Up, ESC, DN, Up, ESC, Up, ENTER

Then go to the dealer menu:

SUB MENUS - > DEALER - > DISPLAY, then scroll until you see Firmware Version.

skipfire, 2020-05-22:
Unfortunately that sequence isn't letting me in.  Maybe it changed with 1.10?

jgyates, 2020-05-22:
Try this one: UP UP ESC DOWN UP ESC UP
from the top menu selection (SYSTEM, DATE/TIME,BATTERY, SUB-MENUS)

skipfire, 2020-05-22:
And now it's not letting me get to the top with rpm sense loss and error 1505 and won't manually start either.  I had hooked up the motorcycle usb adapter and when I did the display reset.  I've put everything back as it was and I called my service tech to sort out the current error.

liltux, 2020-05-22:
1505 with crank (engine actually turn over?) or 1505 and no crank (engine did not turn over?).  IF the 7.5 amp fuse is out on the top of the controller there will be no output to the StartCrankRelay.  Just a quick suggestion.

liltux, 2020-05-22:
@jgyates the first button sequence was correct but you do have to be on the default or top menu with Status Date/time
SetupWifi Submenus
when pressing in the button sequence.  Also if done correctly you should end up in the date/time option.  Common problem is people get in a hurry with the sequence and the controller does not register the button press or a double press is register as one and will not accept the entry.

liltux, 2020-05-22:
@skipfire to clear alarm messages, press OFF then Enter.  IF the condition is not present then the fault will clear.

swvalenti194, 2020-05-22:
I had this issue before. The molex plug in front of the wifi module plug. Long skinny one in your photo wasn't properly seated. Make sure it's plugged in fully seated.

skipfire, 2020-05-22:
@liltux I didn't want to pollute this thread with unrelated stuff, but it was no turn over though I did see some mechanism move.  I'll check the fuse a bit later, maybe once it cools off a bit (over 90 outside right now).  When/if I get past that I will try getting into the dealer menu again to check the version.  @swvalenti194 thanks for that suggestion as well, I'll also check that.

The bigger concern I have is how hooking up the motorcycle USB adapter could have caused that.  https://www.amazon.com/gp/product/B07WXKPTS3/ref=ppx_yo_dt_b_asin_title_o00_s00?ie=UTF8&psc=1
I had pulled the fuse on the adapter before hooking it in, and when I plugged in the USB adapter part (fuse still out) is when it seemed to have a bit of a hard reset.  The only thing that should have been connected at that point was the ground and the volt meter on the USB did not light up, but then once I put the fuse into the USB adapter, it lit up with 13.0v right away.

I will come back with the version information when I get into the menus.

jwtaylor310, 2020-05-22:
Skipfire:

 

CRC errors are probably not the result of the firmware change.  My generator got the update several days ago and Genmon is showing the following: 

 

  Communication Stats :
        Packet Count : M: 3361664, S: 3361663

        CRC Errors : 0 

        CRC Percent Errors : 0.00%

        Timeout Errors : 0

        Timeout Percent Errors : 0.00%

        Modbus Exceptions : 3361663

        Validation Errors : 0

        Invalid Data : 0

        Discarded Bytes : 0

        Comm Restarts : 0

        Packets Per Second : 60.84

        Average Transaction Time : 0.0000 sec

 

Node that the “Packet Count” statistics match the Modbus Exception” count and there are no CRC or other errors.  That indicates that communications between the generator and the Pi are working correctly (no CRC, Timeout or Invalid Data errors) but because of the encryption the contents of the packets is in a ‘foreign language’ so Genmon can’t process them (Modbus Exceptions).

 

--John

jgyates, 2020-05-22:
@skipfire, Let me know once you get the firmware version info. I looked at your log files. the response from the controller is different from response seen from @jwtaylor310. At this point it could be the controller encryption, but the only really way to know for sure is the version would be V1.10.

jwtaylor310, 2020-05-22:
Skipfire:

 

If you have everything wired as originally and you are now getting the 1505 error it is possible that you scrambled the controller’s memory (hopefully only temporarily).  If your tech hasn’t already arrived, you could try power cycling (resetting) the controller.  To do so, remove the 7.5 Amp fuse on the top of the controller and then disconnect the small 2 wire white amphenol connector beneath the controller chassis.  You will need to remove the cover on the right side of the generator to get access to the amphenol connector.  With the fuse out and the 2 wire cable disconnected, the controller’s display should go blank.  Then wait 30 seconds and reconnect.  The controller should go through an initialization process and reconnect to your WiFi.  If it doesn’t, that’s bad news….

 

Good luck!

skipfire, 2020-05-22:
@liltux it was the fuse, thanks.  A trip to a couple auto stores and I found one, replaced, pressed off and enter, then manual and it started right up.

Version is 1.09

@jgyates since it is 1.09 and thus should not be an encryption issue, I'm a bit unsure but I will run through all the other serial port debugging information on the site before asking any other questions.

All who helped - thanks for the assistance so far.  I think I'm going to try powering the Pi off a battery pack until I get everything sorted out so I don't have to touch the generator's battery cables multiple times.  I do have pin-2 ground tied into the serial cable so I should not have a ground issue.

grsthegreat, 2020-05-22:
Why did you insert a motorcycle usb adapter into the USB port. This is not a powered port. Its for reading a flash drive only.

skipfire, 2020-05-22:
@grsthegreat I didn't, the motorcycle adapter was connected to the generator battery.  The motorcycle adapter doesn't have a male port to be able to plug into the USB on the controller.

grsthegreat, 2020-05-22:
oh...i see. i misunderstood the post. my mistake.

jgyates, 2020-05-30:
I am closing this issue since we have another thread on the same topic (https://github.com/jgyates/genmon/issues/374)

closed 2020-05-30 by jgyates